PX4/PX4-Autopilot #11640: PX4FMUv5 / auxiliary PWM Pins

michaelpantic, 2019-03-13:
Hi all,
We observed some issues with the auxiliary PWM outputs on the PX4fmu_v5. 
We're using the holybro px4, and the issue shows similar behaviour on all recent-ish supported firmware version (1.8.2, 1.8.0 1.7.4, 1.71).

- We first observed the problem as discussed here:https://github.com/PX4/Firmware/issues/10677 The lower props of a 12 rotor config do not turn on (upper 6 are on the MAIN/IO, lower 6 on the AUX/FMU)

- Startup scripts etc look correct. So when I manually load the mixer file for the upper rotors (`mixer load /dev/pwm_output0 /etc/mixers/forgotfilename.main.mix`), this works. If I try to load the mixer file for the bottom rotors (`mixer load /dev/pwm_output1 /etc/mixers/forgot_bottom.aux.mix`) it throws failed to load mixer. It does that for any arbitrary mixer file, as soon as I want to load it to /dev/pwm_output1

- fmu driver is started and set to  mode_pwm8. Stopping/starting/setting as thread doesn't change behaviour.

- "fmu test" reboots the pixhawk. "fmu fake 100 100 100 100 100 " doesn't do anything

- "pwm -d /dev/pwm_device0" works nicely (IO pwm's). 
- "pwm -d /dev/pwm_device1" reboots the system. 

So long story short, the PWM FMU driver of the px4fmu_v5 seems to have an issue - we can't get it to work. 

Any idea what we're doing wrong or if there's something wrong with the firmware/driver/nuttx board definition? Thanks much!

julianoes, 2019-04-10:
There might have been some fixes recently. Have you tried if it works with master?

Evaneee, 2019-05-27:
have same question,
pwm test -d /dev/pwm_output0 -c 12345678 -p 1200  work well
pwm test -d /dev/pwm_output1 -c 12345678 -p 1200 donot work!!

julianoes, 2019-05-27:
@Evaneee please create a separate issue. I don't think you have the same question.

Evaneee, 2019-05-27:
> @Evaneee please create a separate issue. I don't think you have the same question.

OK

michaelpantic, 2019-05-27:
@julianoes  Thanks for your answer - we havnt had time to reflash in the meantime. But I keep you posted.

Regarding what @Evaneee  posted - that is how the problem manifests itself. pwm_output0 works with all the debug commands, pwm_output1 either does not respond or crashes the system.

julianoes, 2019-05-27:
@Evaneee sorry, I was assuming that your `pwm test -d /dev/pwm_output1` does not work because  the FMU side has only 6 ports on many Pixhawk designs, however, on Pixhawk 4 (px4_fmu-v5) it's indeed 8, so that's my bad.

I will look into it.

julianoes, 2019-05-30:
Sorry, I did not yet get to this but it's on my list.

julianoes, 2019-06-05:
@michaelpantic I can reproduce the issue that the test output does not work, however, I have not seen the system crashing yet. Could you give me exact steps to reproduce the crash?

closed 2019-09-03 by julianoes

Mario2410, 2019-09-10:
Hi, also have the same problem. is there any update right now?
thanks a lot :)