PX4/PX4-Autopilot #11644: X-Plane SITL

SafatKhan, 2019-03-14:
**This is my first feature request ever, so please pardon my mistakes if I am doing something wrong.**

I have been successful at setting up a HITL simulation using Pixhawk 4, PX4, QGroundControl and X-Plane 10. The problem is that the documentation for SITL states that X-Plane won't be supported in SITL mode. 

Not only is X-Plane very popular, it also has a very respectable simulation fidelity. It even provides the tools to design your own aircraft. That is an amazing feature to have for people like us who are looking to build/test our own models. 

Considering how PX SITL works with 3 other simulators (Gazebo, JMavSim & AirSim), why can't we have it work with X-Plane?

mhkabir, 2019-03-14:
If you're willing to develop this feature and send a pull request, we would be very happy to bring it to upstream.

The main reason this doesn't exist yet is probably because no one else had this use case before.

SafatKhan, 2019-03-14:
> If you're willing to develop this feature and send a pull request, we would be very happy to bring it to upstream.
> 
> The main reason this doesn't exist yet is probably because no one else had this use case before.

I am still trying to learn/comprehend the PX4 code. Not to mention my coding skills, specifically c++, are terrible. 

But truth be told, the greater challenge at this point is understanding how the PX4 SITL works under the hood. I've had an exchange about this issue with Julian Oes about this matter to try and understand why the X-Plane SITL was left out of the PX4 SITL. I am still not sure why what he said matters. Here's the link to that thread for anyone who is interested : https://px4.slack.com/archives/C629RE9H7/p1552487784067600

If I understood Julian's explanation correctly, then it looks like it should be possible to to alter the SITL code in a way such that it just takes the data stream from X-Plane (like in HITL mode) instead of attempting to simulate the sensors itself.

hamishwillee, 2019-03-14:
> Considering how PX SITL works with 3 other simulators (Gazebo, JMavSim & AirSim), why can't we have it work with X-Plane?

> to try and understand why the X-Plane SITL was left out of the PX4 SITL

PX4 is an open source project. Developers who want functionality create and contribute that functionality for everyone to share. So it is not a matter that "X-Plane support was left out", but more that no one has yet decided they need the support enough to develop it. As an example, the PX4 dev team did not and do not "support Airsim" - this is something that Microsoft developed and made available. 
We do welcome feature requests, because sometimes the dev team or broader community will decide that the feature is needed and put their time into the work.

julianoes, 2019-03-15:
I remember that a couple of years back X-Plane was only able to simulate at the attitude level, not at the sensor level. However, I might remember this all wrong or it might have changed by now, so the first step would be to figure out the interface on how to talk to X-Plane.

Starting of HITL used to be connected with QGC but I don't think that's supported by QGC anymore or something we want to build. A scripted startup like for Gazebo or jMAVSim is simpler anyway.

hamishwillee, 2019-03-18:
Thanks @julianoes 

> I remember that a couple of years back X-Plane was only able to simulate at the attitude level, not at the sensor level. However, I might remember this all wrong or it might have changed by now, so the first step would be to figure out the interface on how to talk to X-Plane.

From what I'm hearing Safat is not sure he can do this, and I doubt you have bandwidth. Do you know anyone who has interest in maintaining/updating this?

> Starting of HITL used to be connected with QGC but I don't think that's supported by QGC anymore or something we want to build. A scripted startup like for Gazebo or jMAVSim is simpler anyway.

The docs at least say that XPlane connection is still done via QGC: http://dev.px4.io/en/simulation/hitl.html#x-plane-hitl-environment. Those were produced and tested in just over a year ago (with aid of @dagar) so I suspect it has not been changed. 

BUT I strongly agree a consistent scripted setup approach for XPlane too would be much better, bringing it into line with Gazebo and JMavsim

moreba1, 2021-01-05:
@SafatKhan 
Hi
Is it possible to set the wind in x-plane for simulation?

SafatKhan, 2021-01-05:
@moreba1 

Yes. If you are talking about supplying it customized wind/weather data instead of whatever it may use as default, then yes, it is possible. Please read this : https://developer.x-plane.com/article/custom-wind-weather-in-x-plane-10/