PX4/PX4-Autopilot #11645: Matlab px4 firmware bug on build

alexisutc, 2019-03-14:
Hello all,

i go threw following topic
[Pixhawk Support Package](https://fr.mathworks.com/hardware-support/forms/pixhawk-downloads-conf.html?elqsid=1552487599896&potential_use=Education)

finally when i try to run the make 
make px4fmu-v4_default

command, it stops with following line 

> stdio/lib_sscanf.c:419:28: error: this statement may fall through [-Werror=implicit-fallthrough=]
                       sign = true;
                            ^
stdio/lib_sscanf.c:420:21: note: here
                     case 'u':
                     ^~~~
cc1: all warnings being treated as errors
make[5]: *** [bin/lib_sscanf.o] Error 1
make[4]: *** [libc/libc.a] Error 2
CMakeFiles/nuttx_export_px4fmu-v2.dir/build.make:60: recipe for target 'px4fmu-v2/nuttx_export.stamp' failed
make[3]: *** [px4fmu-v2/nuttx_export.stamp] Error 2
CMakeFiles/Makefile2:1878: recipe for target 'CMakeFiles/nuttx_export_px4fmu-v2.dir/all' failed
make[2]: *** [CMakeFiles/nuttx_export_px4fmu-v2.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:164: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2

Can anybody help me solving this issue please ?

SafatKhan, 2019-03-15:
I think the PX4 team will take no responsibility here. You probably have to contact the PSP team at MathWorks. I am trying to understand this tool myself but am not fully sure what's the use. The guide states confusing things that I find to be contradictory at times. I probably have been entirely misled by the marketing of it on the MathWorks page where it was introduced.

alexisutc, 2019-03-15:
Hi,

actually I found the problem : the file windows_bash_nuttx.sh makes the issue...
If you don't use the GCC / gcc-arm-none-eabi v5.4 it doesn't work 
So basically i had to merge the mathworks windows_bash_nuttx.sh file with the file here
[bash on windows](https://dev.px4.io/en/setup/dev_env_windows_bash_on_win.html)

it helped a lot but the systems still doesn't build the firmware of mathworks

SafatKhan, 2019-03-15:
Congrats on fixing one problem and sorry that you immediately ran into another :P 

What OS and Matlab version are you using?

alexisutc, 2019-03-15:
Thanks haha

Matlab 2018b
Windows 10

are you trying also to run the pixhawk with matlab in one of your projects ?

SafatKhan, 2019-03-15:
Here's the problem:

1. You are not using the version of Matlab that they tested this thing on. They say that it works with R2017a & R2017b and that it also works in R2018a with limitations to External Mode functionality. They made no mention of R2018b at all. The thing is, I've discovered from talking to others who have been trying it that it doesn't even fully work with R2017a. What I've heard from other users is that the target-specific simulink files (like the example files) will not open all the features and blocks in simulink 2017a. They will only open completely in 2017b, even though matlab says on their website that the PsP is valid for both 2017a and 2017b.

2. Windows 10 is a malware, so please get rid of it.

Just kidding about the second one. Don't take it seriously :p

Yes, I am trying to use the PSP for my project. My journey went something like this:
- Got my Windows 7 PC ready with Matlab R2017a.
- Learnt that PSP didn't support R2017a or b at the time.
- Waited and they released the R2017 a & b PSP.
- I was overjoyed and tried to install it and failed. 
- Discovered that unlike the previous versions that worked with cygwin on Windows, they now redesigned things to work only with bash on Widows. Which translates to "Windows 10 only"
- Also got in touch with some other users of the PSP and learnt about the R2017a incompatibility issues.
- So I disabled my previous Matlab and got a new license for Linux (Ubuntu 18.04 LTS) and downloaded R2017b. 

That's my journey so far.

alexisutc, 2019-03-15:
A big Hmmmm....

Well actually i try to compile the code with a ubuntu environment in windows...
And I have no real problem with matlab so far .... I thinck, i think its more coming from a gcc incompatibility.

got my PX4 folder with the tools installes from the (rebuild) `windows_bash_nuttx.sh` file (that I can give you if you want)
But i can't build the whole firmware of matlab..
If you want we can exchange on a messenger system (like skype or other) before publishing the solutiion here (to keep the post clean ^^)

SafatKhan, 2019-03-15:
GitHub doesn't show me any private messaging options. Are you on Slack? If you go the the PX4 slack (https://px4.slack.com), you should be able to look me up and send me messages on there. I'd definitely want to be in touch.

alexisutc, 2019-03-15:
Na i don't have slack
can you add me on skype please alexisoffermannpro

mhkabir, 2019-03-15:
Closing this; We don't provide support for the PSP.

closed 2019-03-15 by mhkabir

alexisutc, 2019-03-15:
sad, i was founding the solution