PX4/PX4-Autopilot #11648: Iris optical flow can not take off in Gazebo simulator

shrit, 2019-03-14:
**Describe the bug**
It is not possible to arm or to takeoff the iris quadcopter equipped with optical flow and a lidar in gazebo.  

**To Reproduce**
1. Start a SITL simulation, change the model type to iris with optical flow.
2. Use the DronecodeSDK offboard_velocity example and start  on UDP port 14540
3. The SDK should detect the quadcopter and the autopilot 

**Expected behavior**
 A message will print on stdout: `Waiting for system to be ready`

**Additional context**
I do not know how exactly the optical flow camera works, I also do not know if it does need gps information. However, since I am starting the quadcopter inside the gazebo, GPS info are provided by default. I can run the normal IRIS quadcopter without any problem.  One point to, add looking at the health information in the SDK source code, either the IMU or the position that is not calibrated.

Last year (in May), I did not have any problem using the IRIS with optical flow. Since then I did not use optical flow. I tried to use it in December last year, I had an accelerometer bias and I was not able to take off.

closed 2019-03-26 by shrit

JuSevRo, 2022-02-07:
Hello @shrit ,
I am currently facing the exact same trouble, mi Iris_opt_flow does not lift off, and the offboard _position code gets stucked in `Waiting for system to be ready`.
Seeing that you closed this issue...how did you solve it?

shrit, 2022-02-07:
I think it was a version issue, I really do not remember the exact details of what happened