PX4/PX4-Autopilot #11656: UTM_GLOBAL_POSITION Message for collision avoidance is not send on Pixhawk4 with FW 1.9.0beta

NAmmann, 2019-03-15:
Tying to use the UTM_GLOBAL_POSITION message for collision avoidance with Pixhawk4 and PX4 FW 1.9.0beta. But the MavLink Inspector does not show the message. Not on the telemetry one nor two port.

LorenzMeier, 2019-03-15:
I checked and it is enabled on both channels. So it looks like one of the conditions to send it is not met.

LorenzMeier, 2019-03-15:
Code: https://github.com/PX4/Firmware/blob/0b4751d03167c4ee147f2e91a94f9919baea70e1/src/modules/mavlink/mavlink_messages.cpp#L1692-L1879

Stream config: https://github.com/PX4/Firmware/blob/6672284871d4cd4163e99d68e75831d0495f58c8/src/modules/mavlink/mavlink_main.cpp#L1677

NAmmann, 2019-03-18:
What are the exact conditions to be met for activating the sending of the message?

LorenzMeier, 2019-03-18:
There are no conditions according to the code. I was wondering if your link might be saturated? What is the output of "mavlink status"?

NAmmann, 2019-03-18:
Zur Zeit habe ich eine USB und D2X Verbindung.
[image]

NAmmann, 2019-03-18:
Oh. And this is from a Pixhawk2 Cube. I wanted to check if my problem is just with the Pixhawk 4. And it turns out, that it is not.

NAmmann, 2019-03-18:
Hier nun der Mavlink Status beim Pixhawk 4: [image]
Das zweite Device ist das D2X, auf welchem die UTM_GLOBAL_POSITION Nachricht relevant ist.

eyeam3, 2019-03-25:
I tested this on the bench and receive the UTM_GLOBAL_POSITION on telem1 and telem2. What version of QGC do you use when running Mavlink Inspector?

dk7xe, 2019-03-25:
https://github.com/PX4/Firmware/issues/11718

dk7xe, 2019-03-25:
Similar issue on NXP_FMUK66_V3 #11718

NAmmann, 2019-04-02:
Updating the QGC to a version understanding the message and setting MAV_PROTO_VER = 2 solves the problem.

closed 2019-04-02 by NAmmann