PX4/PX4-Autopilot #11661: Why not use airspeed sp for gain scaling if airspeed disabled?

CarlOlsson, 2019-03-15:
Currently we are scaling the actuators with the trim airspeed (a constant value) if airspeed is disabled in the fixed wing attitude controller
https://github.com/PX4/Firmware/blob/b54ca67de9b5900c7ed11bef323c7b4e5a0339b6/src/modules/fw_att_control/FixedwingAttitudeControl.cpp#L480-L481

It seems sub optimal since the demanded airspeed can be significantly higher if the fixed wing position controller requests it due to ground speed undershoot in high winds. Is there a reason the scaling is not done using the demanded airspeed instead?

LorenzMeier, 2019-03-15:
No, I think that's just historical. I think a test with demanded airspeed would be worthwhile.

Antiheavy, 2019-03-18:
Sorry for my ignorance, but how can airspeed demand be reasonably estimated if airspeed is disabled in the controller?

I asked a related question in this other thread: https://github.com/PX4/Firmware/issues/11677

CarlOlsson, 2019-03-19:
> how can airspeed demand be reasonably estimated if airspeed is disabled in the controller?

For some vehicles a reasonable estimate of airspeed can be acquired from the throttle setting scaled with air density and voltage

Antiheavy, 2019-10-07:
Still might be interesting to look into this.  FYI @Kjkinney