PX4/PX4-Autopilot #11662: error loading <rosparam> tag in px4_config.yaml

senthilarul, 2019-03-15:
We use GitHub issues only to discuss PX4 bugs and new features. For
questions about using PX4 or related components, please use [PX4 Discuss](http://discuss.px4.io/).

Thanks!

Dines97, 2019-03-20:
I have the same issue with you, I also made a clean ubuntu installation. In addition, I can say that I initially used ubuntu 16.04.5 (Everything worked here) and was updated to 16.04.6 (And here it is not)

AnnaMiG, 2019-03-21:
I am having the same issue. Does anybody know why this is happening?

tmarkmann, 2019-03-21:
I solved the issue by downgrading PyYAML from 5.1 to 3.13 manually with "pip uninstall PyYAML" and "pip install PyYAML==3.13"

mrivi, 2019-04-02:
I can confirm that downgrading PyYAML from 5.1 to 3.13 works

julianoes, 2019-04-03:
I ran into this today as well. What is this exclamation mark for?

bys1123, 2019-04-03:
[image]
I have the same issue.

julianoes, 2019-04-03:
For now a workaround is to change:
```
diff --git a/mavros/launch/px4_config.yaml b/mavros/launch/px4_config.yaml
index d199f17..a3edcce 100644
--- a/mavros/launch/px4_config.yaml
+++ b/mavros/launch/px4_config.yaml
@@ -60,7 +60,7 @@ imu:
   frame_id: "base_link"
   # need find actual values
   linear_acceleration_stdev: 0.0003
-  angular_velocity_stdev: !degrees 0.02
+  angular_velocity_stdev: "!degrees 0.02"
   orientation_stdev: 1.0
   magnetic_stdev: 0.0
 
@@ -211,7 +211,7 @@ odometry:
 # px4flow
 px4flow:
   frame_id: "px4flow"
-  ranger_fov: !degrees 6.8  # 6.8 degreens at 5 meters, 31 degrees at 1 meter
+  ranger_fov: "!degrees 6.8"  # 6.8 degreens at 5 meters, 31 degrees at 1 meter
   ranger_min_range: 0.3     # meters
   ranger_max_range: 5.0     # meters
```

but I assume that changes behavior and is not the right fix.

jsalvador4, 2019-04-09:
I also have a similar error, is there any solution to fix in this case?
[image]

julianoes, 2019-04-09:
@jsalvador4 see my answer above.

julianoes, 2019-04-09:
Check what's causing your issue and see what xan fix it.

w0rt4, 2019-04-09:
https://github.com/mavlink/mavros/commit/1e9d805632eb6133c46859104f0fcec2778accdc#diff-1081d05fe29e8c7f51693863fe64cd9f
@jsalvador4 this should fix your issue (in my case i have exactly the same error and it helps)

jsalvador4, 2019-04-10:
@w0rt4 Thanks, worked fine!

closed 2019-10-08 by julianoes