PX4/PX4-Autopilot #11664: Move servo output drivers to actuator_outputs subscribers

bgat, 2019-03-15:
**What is the proposal?**

*Our actuator drivers are all doing the same, redundant work.*

I propose that we generalize the redundant bits, so as to:
* reduce the size and functionality required of actuator hardware drivers;
* make signal conversions more uniform, explicit and intuitive;
* make it easier to add new actuator device types; and,
* make the overall system more configurable at runtime.

**What motivates this proposal?**

Today's actuator output drivers each perform the `actuator_controls->mix->pwm_limit_calc->actuator_outputs` workflow themselves, which is fine when one driver is responsible for all of the vehicle's actuator output signals. Larger and more-sophisticated vehicles increasingly use a diversity of actuator signal types (CAN, PWM, ethernet, carrier pigeon, etc.), which makes them very difficult to control using PX4 without significant code and structure changes.

**How does this proposal help?**

An incremental step towards fixing the problems of the current situation is to move the `actuator_controls->mix stage` of the signal workflow to a common function, which publishes the results to `actuator_outputs` in the same [-1.0, 1.0] scaling strategy as `actuator_controls`. Actuator signal drivers like `snapdragon_pwm_out.c` would then merely subscribe to `actuator_outputs`, convert the values to their native units, and then adjust their hardware according to the results.

**What's the catch?**

Regarding those "native units"...

My proposal doesn't make it any easier to accommodate Multishot, etc. protocols without modifying PWM drivers themselves; I have ideas on that too, but they need to be described elsewhere.

My immediate need is to accommodate a positioner that uses its own, CAN-based protocol. It uses values in the range [0,4095] for full-scale deflection, which is such a trivial conversion that it doesn't justify a completely new "driver" in my mind. While investigating adding support for this actuator, however, I discovered that the current PX4 architecture would still require me to perform servo mixing in my code...which doesn't really make sense to me, because I'm not changing anything about the vehicle other than some of its electronics. Hence this proposal.

**Anything else you want to add while you have the mic?**

I'm glad you asked! :-)

Mathematically-speaking, all of PX4's inner control loops produce target Euler angle rates (roll, pitch, yaw), expressed as fractions of the vehicle's capabilities in the range [-1.0, 1.0]. That is, an output of [-0.75, 0.0, 0.0], means "assert a roll to the left at 75% of the vehicle's total control authority, and neutralize the assertions of pitch and yaw." In the case of a simple, fixed-wing vehicle, this translates roughly to, "move the left aileron down and the right aileron up, both to 3/4 of their total deflection limits."

(Yes, I know that there are other things in `actuator_controls`. I'll get to them someday as well, but for now I think we can leave them as they are.)

For V-tail planes, flying wings, helicopters, quads, UFOs, etc., Euler's coordinate system doesn't map so neatly to the vehicle's onboard actuators. To deal with that, PX4 uses "mixers" which, at least in theory, are just matrix math to transform from one coordinate system (Euler) to another (the geometry of the vehicle's actual end-effector system). So I say, let's just implement it that way: as a matrix math pipeline.

But even on simple vehicles, PX4's mixers are used to do non-coordinate-system-transformation things like shorten servo throws, reverse directions, and other important stuff. While we can certainly bake those features into the same matrix multiplication step we use for mixing, I think it should be a separate stage: the mixing strategy for a V-tail airplane is always the same math, regardless of which way the servos move on the actual vehicle. So let's have a mixer matrix we can point to that says, "if you are flying a V-tail, use this". Then, have another matrix that says, "...and if your servos are wired backwards, add this".

No matter how I look at it, the key to making any of the above work seems to include making today's actuator signal drivers do a lot less work than they are now. Hence, all this verbage.

dagar, 2019-03-18:
I agree this area definitely needs some work.

Complications that need to be factored in.
 - px4io mixing happens separately
 - Mixing combined with the actual "output module" simplifies the mapping a bit
 - As we grow up we'll need to actually use units where possible. An additional normalized abstraction layer might end up holding us back long term.
 - The multicopter mixers are already on their way to becoming more complicated and generalized (6dof). See https://github.com/PX4/Firmware/pull/8063 and https://github.com/jlecoeur/servo_mix_matrix/blob/master/servo_mix_matrix.ipynb.

Although not quite as interesting as a larger architectural cleanup, I would propose the following short term cleanup to help us get a handle on code sprawl.

 * linux_pwm_out
    * bbblue_pwm_rc
    * bbblue_pwm_rc
    * bbblue_pwm_rc

 * px4fmu
 * snapdragon_pwm_out

bgat, 2019-03-19:
I agree that cleaning up all of the drivers you mention would be beneficial all by itself.

jlecoeur, 2019-06-25:
The output modules still need cleanup.

bkueng, 2022-10-25:
I think we did all of this

closed 2022-10-25 by bkueng