PX4/PX4-Autopilot #11670: mpu9250 FIFO overflow

xhy1286, 2019-03-17:
I am using px4V1.8.2 on a raspberry, but when I opened the hardware serial port of the raspberry, it immediately showed up "FIFO overflow",and Accel #0 fail，and then I used TTL to USB,but I found when more than 10 min passed, it showed"FIFO overflow" finally....😭

Jaeyoung-Lim, 2019-03-23:
@xhy1286 You need to give more information. Which raspberry pi are you using? Which baudrate are you using on the rpi side and which telemetry port is it connected to?

xhy1286, 2019-03-23:
@Jaeyoung-Lim I am using rpi3B+,the baudrate is 115200,and I intended to connect it to /dev/ttyAMA0,but it occurred  "FIFO overflow" immediately.So I tried to use TTL to USB to connect it to ttyUSB2,but more than 10 minutes passed, it finally occurred "FIFO overflow".

xhy1286, 2019-03-27:
I found that it happened probably because I did not build the real time kernel to run PX4.....

dagar, 2019-06-25:
Closing this as the raspberry pi will soon be using the in tree mpu9250 driver (https://github.com/PX4/Firmware/pull/12136).

closed 2019-06-25 by dagar