PX4/PX4-Autopilot #11677: Servo output damping at low airspeeds

LorenzMeier, 2019-03-18:
In current master the outputs (without airspeed sensor) seem to get scaled linearly (or something else is going on at low airspeed). Instead there should probably be a minimum / maximum that assumes min airspeed or alike. Note that this might be related to this issue: https://github.com/PX4/Firmware/issues/11661 - which is something different, but addressing both issues consistently might also solve this. Plot to illustrate:

[image]

sanderux, 2019-03-18:
AFAIK it always assumes cruise airspeed param. Any scaling is based on voltage / baro pressure

Antiheavy, 2019-03-18:
>  the outputs (without airspeed sensor) seem to get scaled linearly (or something else is going on at low airspeed).

without an airspeed sensor, how does the system estimate how it should scale the control outputs?

If any new system is developed to damp servo outputs at low speeds, some consideration should go towards systems that do in fact want highly aggressive outputs at low speeds, for example fixed wings which are hand launched and/or deep stall land.  (tangentially related: we've been thinking about how to let the gain scaling continue below min airspeed for more control in critical times of takeoff and landing when airspeed may drop below minimum)

RomanBapst, 2019-03-18:
@Antiheavy 

> without an airspeed sensor, how does the system estimate how it should scale the control outputs?

In fixed wing flight it assumes trim airspeed, in multicoper mode it assumes minimum fixed wing speed.

> If any new system is developed to damp servo outputs at low speeds, some consideration should go towards systems that do in fact want highly aggressive outputs at low speeds, for example fixed wings [...]

This is in fact what's happening right now. We figured that during hover mode the elevons can help quite a bit to stabilize the drone and help fight wind (talking VTOL here).
Have a look at this function here:
https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/FixedwingAttitudeControl.cpp#L469

The problem in the upper log was a mechanical problem that induced massive vibrations into the gyro signal. I'm open to ideas regarding airspeed scaling. The trickiest part is during VTOL transitions when you have both multicopter and fixed wing rate controller active. If both controllers are tuned very well then you'll probably see oscillations.

sanderux, 2019-03-18:
Here is a log of a combined longer flight on a fully stable vehicle demonstrating the same behavior:
https://review.px4.io/plot_app?log=5ff592e4-c425-40c8-8f0e-2309598f12b9

AuterionWrikeBot, 2019-06-11:
➤ Roman Bapst commented:

https://github.com/PX4/Firmware/pull/11897

closed 2019-06-11 by AuterionWrikeBot