PX4/PX4-Autopilot #11683: False ground contact indication can lead mid-air trust to go to 0.

BazookaJoe1900, 2019-03-20:
This is continuation of 
Slow landing improvement suggestion #11172
 
in that pr it was advised to to turn off the motors on the stage of ground-contact only. but to do it in later stage.

> disabling the vertical thrust (setpoint.thrust[2] = 0.0f;) already in the first land detection stage (ground contact). This brings the thrust down and hence further detection stages a lot quicker but in case of false positives gnerates more altitude loss in the time of wrong information. Based on past experience where the controller "by accident" did exactly that but it worked fine.

is there a reason why this aspect of the pr has not been done?

BazookaJoe1900, 2019-06-28:
I think The issue is still relevant

BazookaJoe1900, 2019-09-26:
Still relevant!

LorenzMeier, 2019-10-11:
The problem is that if you do not shut the motors off on ground contact (or at least reduce throttle dramatically) then the resilient vibration and movement of the platform can prevent the ground contact detection to trigger at all.

LorenzMeier, 2019-10-11:
What is important to understand is that the land detector and auto-disarm really only work if configured correctly for an airframe. And it requires a minimum land speed. If you want to land really slow then the correct solution is to have a sonar / lidar to assist. You can do only so much without a proper ground distance sensor.

BazookaJoe1900, 2019-10-12:
@LorenzMeier Thanks,
as I see it, now there is the chance of false ground contact, on-air. that happens to me from time to time, but luckily its short drops and the vehicle restores from that. its sounds that @ArminStrobel doesn't has this luck (#13124)
I did changed the logic on my land detector to reduce the chance that it can happen only when there is landing command. and not just when hovering.
I added [branch <ground_contact_landing>](https://github.com/BazookaJoe1900/Firmware/tree/ground_contact_landing)to show what I did, its based on v1.9.2 and not master for now.

Basically if bit0 of the new parameter LNDMC_LND_TERMS is set, the drone can be at ground_contact_state if its on manual mode or in SETPOINT_TYPE_LAND.
also the time that ground contact need to be detect is now parameter and not hardcoded 350ms

I understand that the code can and should be changed and be more robust, but that works for me... if needed I can work a bit on that so it can be merged to more common uses.

What I think that need to be changed:
1) some of the logic parameters can't be user defined, such "thrust value below 0.3 of the thrust_range (thrust_hover - thrust_min)" is one condition for ground contact. I think it might be more clear, and to have the option to the user. It might be a "difficult" parameter to set, so its hard to define how to handle that... and its critical too.
2) there is no use of AGL data, if exist. its one of the best condition to use.
3) the trust should get lowered gradually, and not as step.

What I am not sure if should be changed:
4) should the trust should reduced on ground contact mode, or later. 
5) should we add the option to disable ground contact at all?

LorenzMeier, 2019-10-12:
I don't have a conclusive assessment yet, but three early feedback comments:

1) We should allow the user to entirely disable this (based on COM_DISARM_LAND) and leave it disabled by default.
2) The trust reduction should be done to non-zero, as motors have a delay spooling back up to idle thrust - so we need to keep them above
3) We will probably be able to fix this for good with the upcoming hover thrust estimator, as then users don't need to configure things any more.

I'm currently discussing this with @MaEtUgR and will come back with a more conclusive answer.

MikeCharikov, 2019-12-03:
Hi! 

setpoint.thrust[0] = setpoint.thrust[1] = setpoint.thrust[2] = 0.0f;

This is how it looks like in our case: [UAV was in RTL state, it tried to land with no success because ground contact was periodically detected.]

https://www.youtube.com/watch?v=x9_WVOkJ-Lo

It is F450 DJI Kit with standard parameters, custom firmware based on 1.9.2. Actually it took some time for us to find the cause of the problem. :)

Also there is difference in implementation and Land Detector Manual:

> Ground Contact
> This state is reached if following conditions are true for 0.35 seconds:
> no vertical movement (LNDMC_Z_VEL_MAX)
> no horizontal movement (LNDMC_XY_VEL_MAX)
> low thrust MPC_THR_MIN + (MPC_THR_HOVER - MPC_THR_MIN) * 0.3 or velocity setpoint is 0.9 of land speed but vehicle has no vertical movement.
> If the vehicle is in position- or velocity-control and ground contact was detected, the position controller **will set the thrust vector along the body x-y-axis to zero**.

Here are the logs: https://logs.px4.io/plot_app?log=25158f04-ce98-47e2-baaa-ab0b2db96dc8

[image]

dagar, 2020-08-02:
MC land detector `ground_contact` detection fixes have been merged in https://github.com/PX4/Firmware/pull/15083.

closed 2020-08-02 by dagar