PX4/PX4-Autopilot #11685: Propellers small rotation during rebooting from QGC on copter

fpvaspassion, 2019-03-20:
**Describe the bug**
Hello,  when changing some parameters like EKF2_AID_MASK we need to reboot controller from QGC. In case of Safety switch in state ARMED propellers are rotating for small time( less than one turn)

**To Reproduce**
Steps to reproduce the behavior:
1. Switch drone into 'ARMING' State using Safety switch button
2. In QGC go to Params->Tools->Reboot
3. Wait until controller starts to boot
4. Propellers will turn for small period of time( corresponds to ~30 degrees)

**Expected behavior**
As proposed solution - is to disarm copter during execution of reboot cmd or disarm during booting 
- Generic quadrotor with Firmware : stable v1.8.2

dagar, 2019-03-20:
Could you try again with recent master? This is one possibility https://github.com/PX4/Firmware/pull/11453.

fpvaspassion, 2019-03-20:
Dagar, with latest master it is ok - no spinning at all on reboot.