PX4/PX4-Autopilot #11687: No Serial Port Setup in updated firmware.

mattgorski, 2019-03-20:
I am trying to setup S. Port Frsky Telemetry to Serial 4/5 and the documentation is very clear here:

https://docs.px4.io/en/peripherals/serial_configuration.html

BUT on updated firmware I do not see any options to adjust Serial ports.  I need telem 2 for my companion computer and would like to use Serial 4/5 for the Frsky Telem.

Where is the following in updated firmware?

Configuration parameter names follow the pattern \*\_CONFIG or \*\_CFG (QGroundControl only displays the parameters for services/drivers that are present in firmware). At time of writing the current set is: GPS_1_CONFIG, GPS_2_CONFIG, ISBD_CONFIG, MAV_0_CONFIG, MAV_1_CONFIG, MAV_2_CONFIG, RTPS_CONFIG, RTPS_MAV_CONFIG, TEL_FRSKY_CONFIG, TEL_HOTT_CONFIG, SENS_LEDDAR1_CFG, SENS_SF0X_CFG, SENS_TFMINI_CFG, SENS_ULAND_CFG.

Select the Serial tab <- I do not see any options here at all for adjusting Serial 4/5 to FrSky

dagar, 2019-03-20:
Which version of PX4 and QGC? At the moment you'll need to be on the development version of PX4 (that's soon going to be stable v1.9.0).

mattgorski, 2019-03-20:
Much appreciation for the super fast response @dagar. I am on stable (1.8.2 I believe).  So I should flash development branch to see these options?

As for QGroundControl I just updated to the most current APP image on my linux laptop and also updated android to current.

mattgorski, 2019-03-20:
Okay so I need https://github.com/PX4/Firmware/releases/tag/v1.9.0-beta1

dagar, 2019-03-20:
Yes you'd need to flash newer PX4 (development or beta) through QGC (to get the updated metadata), then restart QGC. If you still don't see it when QGC is "stuck" on the old metadata that might need to be cleared first.

mattgorski, 2019-03-20:
Perfect thank you so much for the assistance.  I will report back results when I get time to flash tonight.

mattgorski, 2019-03-20:
Okay I can see the options now on 1.9.0 beta but I have another issue.  When I adjust SYS_COMPANION or any other serial setting save and reboot no settings save but sensors calibration went find and saved.  Very odd.  I'll flash back to stable until this is sorted out.  Any pointers would be great @dagar  😃

hamishwillee, 2019-03-20:
@mattgorski I don't understand exactly what you are saying above, but if you're using 1.9.0 you'd use the new serial params **instead** of SYS_COMPANION.

If you want to use 1.8.2 the old instructions were here: https://github.com/PX4/px4_user_guide/blob/f18662cacd2cdd0b3a49699781766894a71ca77b/en/peripherals/frsky_telemetry.md#px4-configuration

mattgorski, 2019-03-21:
Ohh okay so don't use SYS_COMPANION for companion computer link and use just GPS/Serial settings for setting telem2 over a specific port.  So basically I can set Telem 2 to Frsky Telemetry and then also set GPS_2 to Telem 2 for the companion link correct?

mattgorski, 2019-03-21:
Alright so on developer 1.9.0 I now can get Telem 2 (COMPANION LINK BAUD 921600) over telem 2 in serial but now how can I set GPS_2 serial 4/5 to frsky s. port protocol/telem 2?

mattgorski, 2019-03-22:
Okay so now i found TEL_FRSKY_CONFIG and i have tried GPS_2 TELEM/SERIAL 4 and also setting serial GPS_2 to GPS_2 and TELEM/SERIAL 4 and no go.  I bought the cable from craft in theory 

http://www.craftandtheoryllc.com/store/ct-telemetry-cable-for-frsky-radios/

hamishwillee, 2019-03-25:
@mattgorski So just to confirm, 
- you're using master branch and TEL_FRSKY_CONFIG to configure the port on which FrSky runs? 
- you have tried both selecting (and connecting to) ports GPS_2 and TELEM/SERIAL 4, but in neither case things work? YOu also rebooted after making the settings?

skymaze, 2019-04-16:
> Okay so now i found TEL_FRSKY_CONFIG and i have tried GPS_2 TELEM/SERIAL 4 and also setting serial GPS_2 to GPS_2 and TELEM/SERIAL 4 and no go. I bought the cable from craft and theory
> 
> http://www.craftandtheoryllc.com/store/ct-telemetry-cable-for-frsky-radios/

If you are using Pixracer, Set to Disabled and disconnect the USB, then it will work.

mattgorski, 2019-04-16:
Thanks for the reply.  I am actually using Holybro Pixhawk pix32 from
getfpv.  Any assistance helps.

Set what to disabled? TEL_FRSKY_CONFIG? then reconnect?

skymaze, 2019-04-17:
@mattgorski **Only** **Pixracer** should set TEL_FRSKY_CONFIG to disable. It has a FRS port for telemetry, so no need to set TEL_FRSKY_CONFIG. Maybe cable issue?

julianoes, 2019-10-17:
@skymaze has this been resolved?

julianoes, 2020-02-12:
Closing until we get an answer.

closed 2020-02-12 by julianoes