PX4/PX4-Autopilot #11690: The SF11/C is not publish the distance data when the PX4Flow Smart Camera is connected.

bozkurthan, 2019-03-21:
**Describe the bug**
Hello, I'm using SF11/C for distance sensor and PX4Flow Optical Flow Camera in Pixhawk 2.1 (Cube). I don't have an ultrasonic module on PX4Flow. My setup is shown below.

[image]

[image]

Like images PX4Flow Camera is connected to I2C and SF11/C is connected to Telemetry 1.
The bug is revealed when I connected both of sensor to Pixhawk. SF11/C publishes distance sensor when PX4FLow Camera isn't connected. But, When I connect the PX4Flow to Pixhawk, PX4Flow publishes Zero distance which mentioned before this issue [https://github.com/PX4/Flow/issues/101](url). In addition SF11/C cannot communicate Pixhawk when PX4Flow is connected.

Without PX4Flow Camera `sf0x test` and Current Distance Analyze
[image]
[image]

With PX4Flow Camera`sf0x test` and Current Distance Analyze
[image]
[image]

In documentations, Laser Altimeters sensor can be connect instead of Ultrasonic module. But, I didn't find any schematic for this. 

I examined the drivers of PX4Flow and SF0X. They have same definition for Uorb distance_report topic. 

PX4FLOW:
https://github.com/PX4/Firmware/blob/14ef009aabf5723b1a9f8fe5489844c961f05fab/src/drivers/px4flow/px4flow.cpp#L554-L566

SF0X:
https://github.com/PX4/Firmware/blob/14ef009aabf5723b1a9f8fe5489844c961f05fab/src/drivers/distance_sensor/sf0x/sf0x.cpp#L514-L528

Also they have same id for publishing. I tried to give different ID but it didn't work.
Furthermore, I just commented lines on PX4Flow. However, it didn't work as well.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect SF11/C to Pixhawk Telemetry 1 and configure
2. Can be able to see Current Distance
3. Connect PX4FLow to Pixhawk I2C
4. See Current Distance Value Zero

**Expected behavior**
Pixhawk can be able to see correct distance values when PX4Flow and SF11/C are connected.

bozkurthan, 2019-03-22:
@dagar fyi.
I wasn't able to test on I2C bus with same setup (I don't have 1 more I2C connector for this). But today I able to test with poor connection on I2C bus. 
The problem is that PX4Flow and SF11/C must connect on same I2C bus because of PX4Flow driver. So, it can be close. 
Here correct setup:
[image]

[image]

In addition, maybe it can be add for future request. (PX4FLow on I2C, SF11/C on Uart) Should I open new Issue or can we continue on this?

hamishwillee, 2019-03-26:
This also looks related: https://github.com/PX4/Firmware/issues/11227

bozkurthan, 2019-03-27:
As far as I understand, it is not same issue. @60ericleal60 couldn't get any data from PX4Flow camera but I could. I just have problem with distance sensor.

60ericleal60, 2019-03-27:
There is a solder bridge you need to fill to power the distance sensor.

bozkurthan, 2019-03-27:
You are right. I saw that bridge and I tried to connect SF11/C to it with poor connection but data didn't come Qgroundcontrol. I'll try again with solder.

bozkurthan, 2019-03-28:
This issue is fixed at #11724 by @DanielePettenuzzo .  I'm closing and will continue on #11689

closed 2019-03-28 by bozkurthan