PX4/PX4-Autopilot #11691: "tests all" command remove all configuration of Pixhawk

bozkurthan, 2019-03-21:
I'm using Pixhawk 2.1 ( Cube ) with 1.9.0(beta) firmware. I just wanted to test sensor with the nuttx shell "tests" command. However, I saw the option "tests all". Before I used the command, I configured whole sensors, airframe etc.. After used command whole setup configuration is went off. I have parameters backup but it's a real problem for someone who don't have. 
Can someone test also this command and if it's only related to me?

Screenshots:
Before use 'tests all'
[image]

Using command
[image]

After use
[image]

FlavioTonelli, 2019-07-05:
I can reproduce this behavior on my Pixhawk cube as well.
@julianoes just pinging to let you know here the stale bot is on my opinion wrong.

julianoes, 2019-07-05:
I wonder if this is only on Cube or in general.

julianoes, 2019-07-05:
@davids5 FYI

FlavioTonelli, 2019-07-05:
> I wonder if this is only on Cube or in general.
It seems general. Just tested on a Pixhawk3 Pro.

davids5, 2019-07-05:
@bozkurthan @julianoes - the `test all` does a `tests parameters' and it is a destructive test on parameters. 

```
nsh>  param show sys*
Symbols: x = used, + = saved, * = unsaved
x   SYS_AUTOCONFIG [508,1034] : 0
x + SYS_AUTOSTART [509,1035] : 4011
x   SYS_CAL_ACCEL [510,1037] : 0
x   SYS_CAL_BARO [511,1038] : 0
x   SYS_CAL_GYRO [512,1039] : 0
x   SYS_CAL_TDEL [513,1040] : 24
x   SYS_CAL_TMAX [514,1041] : 10
x   SYS_CAL_TMIN [515,1042] : 5
x   SYS_COMPANION [516,1043] : 0
x   SYS_FMU_TASK [517,1044] : 1
x   SYS_FORCE_F7DC [518,1045] : 0
x   SYS_HAS_BARO [519,1046] : 1
x   SYS_HAS_MAG [520,1047] : 1
x   SYS_HITL [521,1048] : 0
x   SYS_MC_EST_GROUP [522,1049] : 2
x   SYS_PARAM_VER [523,1050] : 1
x   SYS_RESTART_TYPE [524,1051] : 2
x   SYS_STCK_EN [525,1052] : 1
x   SYS_USE_IO [526,1053] : 1

 560/1285 parameters used.
nsh>  tests parameters
INFO  [tests] RUNNING TEST: ResetAll
INFO  [tests] TEST PASSED: ResetAll

INFO  [tests] RUNNING TEST: SimpleFind
INFO  [tests] TEST PASSED: SimpleFind

INFO  [tests] RUNNING TEST: ResetAll
INFO  [tests] TEST PASSED: ResetAll

INFO  [tests] RUNNING TEST: ResetAllExcludesOne
INFO  [tests] TEST PASSED: ResetAllExcludesOne

INFO  [tests] RUNNING TEST: ResetAllExcludesTwo
INFO  [tests] TEST PASSED: ResetAllExcludesTwo

INFO  [tests] RUNNING TEST: ResetAllExcludesBoundaryCheck
INFO  [tests] TEST PASSED: ResetAllExcludesBoundaryCheck

INFO  [tests] RUNNING TEST: ResetAllExcludesWildcard
INFO  [tests] TEST PASSED: ResetAllExcludesWildcard

INFO  [tests] RUNNING TEST: exportImport
INFO  [tests] TEST PASSED: exportImport

INFO  [tests] ALL TESTS PASSED
INFO  [tests]   Tests passed :      8
INFO  [tests]   Tests failed :      0
INFO  [tests]   Tested assertions : 42
INFO  [tests] TuneID: 14
parameters PASSED
nsh> INFO  [ekf2] Mag sensor ID changed to 396825
 param show sys*
Symbols: x = used, + = saved, * = unsaved
x   SYS_AUTOCONFIG [508,1034] : 0
x   SYS_AUTOSTART [509,1035] : 0
x   SYS_CAL_ACCEL [510,1037] : 0
x   SYS_CAL_BARO [511,1038] : 0
x   SYS_CAL_GYRO [512,1039] : 0
x   SYS_CAL_TDEL [513,1040] : 24
x   SYS_CAL_TMAX [514,1041] : 10
x   SYS_CAL_TMIN [515,1042] : 5
x   SYS_COMPANION [516,1043] : 0
x   SYS_FMU_TASK [517,1044] : 1
x   SYS_FORCE_F7DC [518,1045] : 0
x   SYS_HAS_BARO [519,1046] : 1
x   SYS_HAS_MAG [520,1047] : 1
x   SYS_HITL [521,1048] : 0
x   SYS_MC_EST_GROUP [522,1049] : 2
x   SYS_PARAM_VER [523,1050] : 1
x   SYS_RESTART_TYPE [524,1051] : 2
x   SYS_STCK_EN [525,1052] : 1
x   SYS_USE_IO [526,1053] : 1

 560/1285 parameters used.
nsh>
```

@hamishwillee - do we need a warning is the docs?

closed 2019-07-05 by davids5

julianoes, 2019-07-08:
@davids5 could we restore the params after the test?

davids5, 2019-07-08:
@julianoes we could and code size and exec time expenses. . But this is not a use case of the system as far as I know. @dagar may have an opinion.

dagar, 2019-07-08:
We could split the destructive part into a standalone test that doesn't run by default. It could also print a huge warning about what's going to happen.