PX4/PX4-Autopilot #11695: Accelerometer calibration problem on FMUV5

RomanBapst, 2019-03-21:
**Describe the bug**
We noticed a high magnitude of the accelerometer Z component which we could not eliminate even after several accelerometer calibrations. It seems that only the first instance of the sensor_accel topic is affected. Furthermore, the problem does not occur with 1.8.2 on the same hardware with the same parameters.

[image]

RomanBapst, 2019-03-21:
@bkueng @dagar FYI

dagar, 2019-03-21:
Were the calibration parameters changing between attempts or relative to 1.8.2?

RomanBapst, 2019-03-21:
@dagar No

RomanBapst, 2019-03-21:
@dagar Sorry, I should explain better. Calibration params were not changed when moving between master and 1.8. We are able to reproduce this on several devices.

julianoes, 2019-03-22:
[image]

Two things that caught my eye:

The X-offset is really high (~2 m/s^2) although this seems to be correct for my Pixhawk 4.
The Z-offset is small but if I change the sign of it, we reach 9.81 much more accurate.

dagar, 2019-03-24:
I reproduced this locally and confirmed both stable (v1.8.2) and current master produce effectively the same calibration parameters on the same hardware.

dagar, 2019-03-24:
The raw data looks equivalent, but the scaling changed.

bresch, 2019-03-24:
@dagar After bisecting the commits, I found the guilty one:

```
commit d299d439c626e2cbdb94d506d857fb7e664cfe18
Author: Daniel Agar <daniel@agar.ca>
Date:   Mon Jan 14 16:15:23 2019 -0500

    mpu6000 use new PX4Accelerometer and PX4Gyroscope classes
```

On commit https://github.com/PX4/Firmware/commit/d299d439c626e2cbdb94d506d857fb7e664cfe18
```
INFO  [sensors] accel status:
INFO  [ecl/validation] validator: best: 1, prev best: 1, failsafe: NO (0 events)
INFO  [ecl/validation] sensor #0, prio: 75, state: OK
INFO  [ecl/validation] 	val:  -0.0356, lp:  -0.0269 mean dev:  -0.0076 RMS:   0.0773 conf:   1.0000
INFO  [ecl/validation] 	val:   0.2122, lp:   0.1995 mean dev:   0.0003 RMS:   0.0184 conf:   1.0000
INFO  [ecl/validation] 	val: -10.3720, lp: -10.3181 mean dev:   0.0002 RMS:   0.0437 conf:   1.0000
INFO  [ecl/validation] sensor #1, prio: 99, state: OK
INFO  [ecl/validation] 	val:  -0.1478, lp:  -0.1488 mean dev:  -0.0033 RMS:   0.0398 conf:   1.0000
INFO  [ecl/validation] 	val:   0.1883, lp:   0.1141 mean dev:  -0.0006 RMS:   0.0232 conf:   1.0000
INFO  [ecl/validation] 	val:  -9.7528, lp:  -9.7770 mean dev:   0.0004 RMS:   0.0521 conf:   1.0000
```
On the previous one (91dcfb7ab2cc56a4dbfcdd4f20e655264e2324dc):
```
INFO  [sensors] accel status:
INFO  [ecl/validation] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [ecl/validation] sensor #0, prio: 100, state: OK
INFO  [ecl/validation] 	val:  -0.0584, lp:  -0.0613 mean dev:  -0.0028 RMS:   0.0488 conf:   1.0000
INFO  [ecl/validation] 	val:   0.0588, lp:   0.0928 mean dev:  -0.0001 RMS:   0.0169 conf:   1.0000
INFO  [ecl/validation] 	val:  -9.7833, lp:  -9.8157 mean dev:   0.0007 RMS:   0.0701 conf:   1.0000
INFO  [ecl/validation] sensor #1, prio: 99, state: OK
INFO  [ecl/validation] 	val:  -0.1555, lp:  -0.1432 mean dev:  -0.0003 RMS:   0.0179 conf:   1.0000
INFO  [ecl/validation] 	val:   0.0595, lp:   0.0846 mean dev:  -0.0001 RMS:   0.0224 conf:   1.0000
INFO  [ecl/validation] 	val:  -9.7305, lp:  -9.7932 mean dev:   0.0001 RMS:   0.0733 conf:   1.0000
```

Note that I didn't do any calibration between the two tests, It's then not a calibration issue.

dagar, 2019-03-24:
The raw data is equivalent between stable (left) and master (right).

[image]

dagar, 2019-03-24:
It's even more apparent looking at the absolute value of the filtered accel data. Again stable left, master right.

[image]

dagar, 2019-03-24:
It looks like this is coming from floating point loss of precision. In PX4 master the sensor data is rotated after being scaled and stored as floats (raw data is int16). In stable the rotation is done with the data still scaled (2048 G/LSB), then scaled down.

dagar, 2019-03-24:
> @dagar After bisecting the commits, I found the guilty one:
> 
> ```
> commit d299d439c626e2cbdb94d506d857fb7e664cfe18
> Author: Daniel Agar <daniel@agar.ca>
> Date:   Mon Jan 14 16:15:23 2019 -0500
> 
>     mpu6000 use new PX4Accelerometer and PX4Gyroscope classes
> ```
> On commit [d299d43](https://github.com/PX4/Firmware/commit/d299d439c626e2cbdb94d506d857fb7e664cfe18)
> 
> ```
> INFO  [sensors] accel status:
> INFO  [ecl/validation] validator: best: 1, prev best: 1, failsafe: NO (0 events)
> INFO  [ecl/validation] sensor #0, prio: 75, state: OK
> INFO  [ecl/validation] 	val:  -0.0356, lp:  -0.0269 mean dev:  -0.0076 RMS:   0.0773 conf:   1.0000
> INFO  [ecl/validation] 	val:   0.2122, lp:   0.1995 mean dev:   0.0003 RMS:   0.0184 conf:   1.0000
> INFO  [ecl/validation] 	val: -10.3720, lp: -10.3181 mean dev:   0.0002 RMS:   0.0437 conf:   1.0000
> INFO  [ecl/validation] sensor #1, prio: 99, state: OK
> INFO  [ecl/validation] 	val:  -0.1478, lp:  -0.1488 mean dev:  -0.0033 RMS:   0.0398 conf:   1.0000
> INFO  [ecl/validation] 	val:   0.1883, lp:   0.1141 mean dev:  -0.0006 RMS:   0.0232 conf:   1.0000
> INFO  [ecl/validation] 	val:  -9.7528, lp:  -9.7770 mean dev:   0.0004 RMS:   0.0521 conf:   1.0000
> ```
> On the previous one ([91dcfb7](https://github.com/PX4/Firmware/commit/91dcfb7ab2cc56a4dbfcdd4f20e655264e2324dc)):
> 
> ```
> INFO  [sensors] accel status:
> INFO  [ecl/validation] validator: best: 0, prev best: 0, failsafe: NO (0 events)
> INFO  [ecl/validation] sensor #0, prio: 100, state: OK
> INFO  [ecl/validation] 	val:  -0.0584, lp:  -0.0613 mean dev:  -0.0028 RMS:   0.0488 conf:   1.0000
> INFO  [ecl/validation] 	val:   0.0588, lp:   0.0928 mean dev:  -0.0001 RMS:   0.0169 conf:   1.0000
> INFO  [ecl/validation] 	val:  -9.7833, lp:  -9.8157 mean dev:   0.0007 RMS:   0.0701 conf:   1.0000
> INFO  [ecl/validation] sensor #1, prio: 99, state: OK
> INFO  [ecl/validation] 	val:  -0.1555, lp:  -0.1432 mean dev:  -0.0003 RMS:   0.0179 conf:   1.0000
> INFO  [ecl/validation] 	val:   0.0595, lp:   0.0846 mean dev:  -0.0001 RMS:   0.0224 conf:   1.0000
> INFO  [ecl/validation] 	val:  -9.7305, lp:  -9.7932 mean dev:   0.0001 RMS:   0.0733 conf:   1.0000
> ```
> Note that I didn't do any calibration between the two tests, It's then not a calibration issue.

Ha, thanks @bresch. I only just saw your reply now.

commit 370ff381 (2019-03-24): PX4Accelerometer apply sensor rotation before scaling

 - prevents loss of numerical precision
 - fixes #11695
PR #11713 (2019-03-24): PX4Accelerometer apply sensor rotation before scaling
 - prevents loss of numerical precision
 - fixes #11695
commit 5a841761 (2019-03-24): PX4Accelerometer apply sensor rotation before scaling

 - prevents loss of numerical precision
 - fixes #11695
closed 2019-03-24 by LorenzMeier