PX4/PX4-Autopilot #11697: Problem with MAVLINK_MSG_ID_HIL_ACTUATOR_CONTROLS

lovettchris, 2019-03-21:
**Repro**: run [AirSim](https://github.com/Microsoft/AirSim) with latest v1.8.2 firmware on Pixhawk v2 in HIL_QUADROCOPTER_X mode.  Try and take off, nothing happens, rotors do not spin.

The PX4 status shows MAV_MODE_FLAG_HIL_ENABLED and AirSim is getting MAVLINK_MSG_ID_HIL_ACTUATOR_CONTROLS , but the actuator control values in the message are all zeros all the time even when my RC radio throttle is set to full.

See my [px4 log file](https://logs.px4.io/plot_app?log=b69e4688-8dd7-43d9-b965-6c2d1bea2aab).  You can see my manual Throttle input in "Manual Control Inputs" graph - but these throttle values are not making it out to the HIL_ACTUATOR_CONTROLS message.

If I move my Pixhawk back to version 1.7.0 of the PX4 Firmware then everything works fine.  Has something changed recently in the mavlink protocol with PX4 in this area?  

Note that if I also connect to nuttx shell the commander seems happy enough, but still sending no actuator controls to AirSim:
```
nsh> commander status
commander status
WARN  [commander] type: symmetric motion
WARN  [commander] safety: USB enabled: [OK], power state valid:  [OK]
WARN  [commander] home: lat = 47.6728307, lon = -122.1229818, alt = NaN, yaw: 0.08
WARN  [commander] home: x = 0.0055329, y = 0.0004450, z = NaN
WARN  [commander] datalink: OK
WARN  [commander] main state: 11
WARN  [commander] nav state: 18
WARN  [commander] arming: STANDBY
nsh> commander arm
commander arm
nsh> STATUS: sev=6, 'ARMED by command line'
STATUS: sev=6, '[logger] file: /fs/microsd/log/2019-03-21/21_00_2'
commander takeoff
commander takeoff
nsh> STATUS: sev=6, 'Using minimum takeoff altitude: 2.50 m'
    PX4_CUSTOM_MAIN_MODE_AUTO
    PX4_CUSTOM_SUB_MODE_AUTO_TAKEOFF
commander land
commander land
nsh> STATUS: sev=6, 'Landing at current position'
com    PX4_CUSTOM_MAIN_MODE_AUTO
    PX4_CUSTOM_SUB_MODE_AUTO_LAND
commander disarm
nsh> STATUS: sev=6, 'DISARMED by command line'
```

px4 parameters: [params.zip](https://github.com/PX4/Firmware/files/2993971/params.zip)

bkueng, 2019-03-25:
Can you flash the v3 target? You might have to update the bootloader first (https://docs.px4.io/en/config/firmware.html#bootloader).

lovettchris, 2019-04-07:
I found the problem.  Turns out **pwm_out_sim** module was removed from nuttx_px4fmu-v2_default.cmake, but this module is needed to generate fake PWM actuator_outputs when running in HITL mode.  Uncommenting this line and rebuilding v1.8.2 firmware works fine.
```
       #drivers/pwm_out_sim
```

dagar, 2019-04-07:
@lovettchris there should now be sufficient flash space to re-enable pwm_out_sim by default.

lovettchris, 2019-04-07:
Yes, there was enough space, I didn't have to remove anything else.

v01d, 2019-07-26:
Hi,
I have 1.9.2. installed and I have the same problem, even though pwm_out_sim is running. Anyway to further debug this?

v01d, 2019-07-30:
I see #12249 might be related (same symptom)

v01d, 2019-08-05:
Also possibly related #12629

julianoes, 2019-08-06:
@lovettchris I'm trying to reproduce the issue but I don't see it with jMAVSim.

I've therefore started to build AirSim on my Linux (Fedora 30) but I ran into other troubles there:

```
+ mono Binaries/DotNET/UnrealBuildTool.exe -projectfiles
Generating data for project indexing... 0%ERROR: This version of the Unreal Engine can only be compiled with clang 5.0, 4.0, 3.9, 3.8, 3.7, 3.6 and 3.5. clang 8.0.0 may not build it - please use a different version.
```