PX4/PX4-Autopilot #11701: Vehicle did not respond to request for parameters

crossa, 2019-03-22:
QGC says vehicle did not respond to request for parameters,can not display full interface when connect to pixhawk via telemetry 
The issue appeared after QGC@3.2.3. px4@1.7.2

I found if both 'Emit heatbeat' and 'Request' start of MAVLink telemetry streams' checked in qgc, the loss rete of packet grower up to 90%

 I try to flash the follwing firmware into my pixhwak

px4 1.8.2
the lastest stable version of nuttx-arducopter for pixhawk
ArduCopter 3.6.7 without nuttx
The results are clear.

If using px4@1.8.2 and nuttx-arducopter, both QGC and APM planner can't get all the parameter via radio telemetry.

When using nuttx-arducopter , APM planner still can't get all the parameter via radio telemetry

But the ArduCopter without nuttx works fine !

What changes had been done since px4@1.7.2?
Is the configration of mavlink correct in the firmware?
Is the lunch order of the script located in  init.d  correct? 

https://github.com/mavlink/qgroundcontrol/issues/7301

@julianoes @LorenzMeier @bkueng @dogmaphobic

dogmaphobic, 2019-03-22:
@crossa How is the Pixhawk connected to QGC?

>I found if both 'Emit heatbeat' and 'Request' start of MAVLink telemetry streams' checked in qgc, the loss rete of packet grower up to 90%

@dagar This sounds like https://github.com/PX4/Firmware/issues/11674

crossa, 2019-03-23:
QGC connect to pixhawk via radio telemtey and the telemetry is a replica of 3dr 915m radio telemetry. In fact, it is just a wireless serial port.

This issue appeared since px4@1.7.2 released.

I  have tried ardupilot without nuttx provided by apm planner and I find it works fine.

The issue just appears when using the firmware provided by qgc via firmware upgrade.

I find the issues only appears when using the firmware contains rtos.

@dogmaphobic dagar@

LorenzMeier, 2019-03-25:
Can you take a picture of the autopilot and radio and upload it here please?

crossa, 2019-03-25:
Ok, here are the pictures.  Was the configuration of mavlink submodule be changed since px4@1.72？

@LorenzMeier 

[image]
[image]
[image]

dagar, 2019-03-25:
More Sik radios without flow control.

crossa, 2019-03-25:
Was the radio telementry without flow control avaliable for the laster statble version of px4 ？

@dagar @LorenzMeier

Bosuwort, 2019-08-27:
I have the same issue. It always happens that the vehicle can recieve the command from mavlink console at first but have no response later, especially when it transitions to mission mode. I have tried different radio telemetry but it did not work.

Bosuwort, 2019-09-10:
> I have the same issue. It always happens that the vehicle can recieve the command from mavlink console at first but have no response later, especially when it transitions to mission mode. I have tried different radio telemetry but it did not work.

I fixed it. My custom module occupied too much operation resource and other commands can't be received by the vehicle. I added usleep(500000) in it and there's no issues any more.

closed 2019-09-10 by LorenzMeier