PX4/PX4-Autopilot #11707: /mavros/vision_pose/pose topic Send uwb data pixhawk for positioning

LShikang, 2019-03-22:
Hello everyone!Can anyone help me?
We are trying to achieve indoor autonomous flight in a drone we are developing. For this, we want to use Ultra-Wide Band (UWB) beacons. We are already able to track the location (not orientation) of an UWB beacon quite accurately. The next step for us is to have the Pixhawk (original Pixhawk, FMUv2, running PX4 v1.8.1) that we are using make use of this location. For this, we have set up a UP board with mavros, which sends the xy-data from the tracked UWB beacon to the Pixhawk over the /mavros/vision_pose/pose topic.

Pixhawk seems to have correctly received this location, as we saw in VISION_POSITION_ESTIMATE.However, we can't reliably LOCAL_POSITION_NED follow the ultra-wideband position: when the drone's ground is stationary, the position is correct, but when we take off (using position control, for safety), the VISION_POSITION_ESTIMATE still receives the correct, but ekf2 fusion Local_position/pose data cannot follow the data published by vision_pose/pose.

Set EKF2_AID_MASK to 8vision position fusion in qgc,
EKF2_HGT_MODE to 2Range sensor, lidar gives height information,
EKF2_EV_DELAY is 23ms,
EKF2_EV_GATE is 10SD
my uwb is 9Hz

It is worth mentioning that the /mavros/vision_pose/pose orientation we published is read from the orientation of the imu/data topic published by px4, and the z-axis value uses the distance data of the lidar pointing to the ground.

We define uwb with the x-axis north and the y-axis east (transformed on mavros). We did a test, the pixhawk head turned north, but the effect was very bad, not hovering in a fixed place (in position mode). I want to know if our uwb data is well integrated into the pixhawk, what exactly causes the pixhawk to not follow the uwb data very well.
Here is the log we did a test：https://logs.px4.io/plot_app?log=5f5cef98-77c8-4bdc-a3d4-19859dfc4d7d

We also tried to adjust the following parameters
EKF2_ACC_NOISE is 0.7,
EKF2_GYR_NOISE is 0.03,
EKF2_MAG_DECL, but the effect is the same

Jaeyoung-Lim, 2019-03-23:
@LShikang Have you tried changing `EKF2_MAG_TYPE` ?

LShikang, 2019-03-23:
> @LShikang Have you tried changing EKF2_MAG_TYPE ?

no,my EKF2_MAG_TYPE is 0(automatic),and What should we set it up to be?h
Here is the log that we did a test today:https://logs.px4.io/plot_app?log=22cd40f6-3879-4826-9020-315a84571538
It can hover at the beginning, but the position offset is gradually increasing, which is a bit like the error is accumulating. What happened to px4?

LShikang, 2019-03-23:
> @LShikang Have you tried changing EKF2_MAG_TYPE ?

No,my EKF2_MAG_TYPE is 0(automatic),and What should we set it up to be?h
Here is the log that we did a test today:https://logs.px4.io/plot_app?log=22cd40f6-3879-4826-9020-315a84571538
It can hover at the beginning, but the position offset is gradually increasing, which is a bit like the error is accumulating. What happened to px4?

LShikang, 2019-03-23:
@MarLi @mhkabir @Jaeyoung-Lim @#1LorenzMeier @dagar @bkueng @davids5 @#5julianoes    @thomasguble @all
Can you give us some advice?

closed 2019-07-08 by stale[bot]

zhaoxiaowei1013, 2019-11-25:
Have you solved the problem？