PX4/PX4-Autopilot #11709: Always get ACCELS INCONSISTENT - CHECK CAL on Pixhawk4 after 1.9.0 beta1

bys1123, 2019-03-23:
**Describe the bug**
I tried master branch today, because of #11694 looks pretty cool. but I got ACCELS INCONSISTENT - CHECK CAL when I tried to arm the copter.

I also tried re-cal and reset parameters, it was not work either.

Then I tried to built 1.9.0beta1, when Pixhawk 4 power on, it shows:
Preflight Fail: High Accelerometer Bias
Preflight advide: ACCELS INCONSISTENT - CHECK CAL

re-cal and reset parameters won't working.

I tried different Pixhawk4, error is the same.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash master firmware on Pixhawks.
2. Press safety switch.
3. Try to arm on radio
4. See error

**Drone (please complete the following information):**
- DIY F450
- Pixhawk 4

bys1123, 2019-03-23:
Looks like issue #11695 is the same problem.

bys1123, 2019-03-23:
Update: It's not real 1.9.0beta1, I didn't roll back submodules successfully.
So, this bug is in one submodule.

dagar, 2019-03-24:
Thanks for reporting @bys1123. I'll use the existing https://github.com/PX4/Firmware/issues/11695 to track the issue. If you have any additional data please attach it there. This is a blocking issue for the v1.9.0 release.

closed 2019-03-24 by dagar