PX4/PX4-Autopilot #11718: UTM_GLOBAL_POSITION only sent in case of MavLink connection to QGC

dk7xe, 2019-03-25:
**Describe the bug**
The UTM_GLOBAL_POSITION message is sent only in case there is active connection to QGC.
Requirement is that the UTM_GLOBAL_POSITION message is broadcasted on any telemetry port independent from an active connection to a ground control station.
Test was executen on both telemetry ports TELEM1 and TELEM2.
On ground PC mavlink-router is running providing UDP stream on port 14550 to QGC
and port 14551 to "msgrcvapp". msgrcvapp is an own terminal app showing reveived MavLink messages.

**To Reproduce**
**Scenario 1 - MavLink via single telemetry radio either on TELEM1 or TELEM2 (433MHz or D2X)**
1. Drone switched on. QGC not started yet.
2. MavLink Heartbeat message received on "msgrcvapp" at ground PC. But no UTM_GLOBAL_POSITION.
3. start QGC and wait until connection to drone established.
4. UTM_GLOBAL_POSITION is broadcasted.

**Scenario 2 - MavLink via two telemetry radios**
433MHz radio connected to TELEM1, D2X radio connected to TELEM2.
1. Drone switched on. QGC not started yet.
2. MavLink Heartbeat message received on "msgrcvapp" at ground PC. But no UTM_GLOBAL_POSITION.
3. start QGC and connection established via D2X radio on TELEM2.
4. UTM_GLOBAL_POSITION is broadcastet via TELEM2.
5. UTM_GLOBAL_POSITION is not broadcasted via TELEM1 still.

**Expected behavior**
UTM_GLOBAL_POSITION message is sent to all TELEM ports independend from a connection to QGC similar to heartbeat message.

**Log Files and Screenshots**
n/a

**Drone (please complete the following information):**
- NXP_FMUK66_V3 running PX4 FW version 1.9.0 80
- 433MHz telemetry on TELEM1 (57k6 Bd).
- D2X communication unit on TELEM2 (set to 115k2 Bd and normal telemetry).

**Additional context**
For a UTM system identification it is required to have the UTM_GLOBAL_POSITION message sent independend from any active telemetry connection.

eyeam3, 2019-03-27:
I am not able to reproduce this issue with my bench setup. I use MAVLink Inspector in QGC to analyze the stream but physically remove the RX pin on the Pixhawk side. Thus, QGC cannot connect properly with the FC and QGC cannot send MAVLink messages to the drone.
Yet, the moment I plug in TELEM1 or TELEM2, I receive the full stream including the UTM_GLOBAL_POSITION. I also tried from boot with the same result, with an FTDI and a 433MHz radio pair. 

Moreover, I tried to reproduce the second scenario and immediately received the message. Connecting QGC over the other TELEM port did not interrupt the first stream and the UTM_GLOBAL_POSITION  was received on both ends.

Could you test your setup without the mavlink-router? I did not have this in my setup. Otherwise, more information would be needed to debug this issue.

LorenzMeier, 2019-03-27:
Sounds very much like a router issue where it detects where to route messages and gets that wrong.

dk7xe, 2019-03-29:
hmm ok.. then my nice video is useless https://youtu.be/_x8DXO6efq4

dk7xe, 2019-03-29:
@eyeam3 Michael, i did another test beside the one documented in video of previous post with the same setup as you suggested with FTDI and TX and GND connected and indeed... **NO** UTM_GLOBAL_POSITION message is reseived!!
So the issue seems to be with the PX4 FW we all are using in the project.

dk7xe, 2019-04-01:
Setting MAV_PROTO_VER = 2 actually solves the problem i have been facing.

closed 2019-04-01 by eyeam3