PX4/PX4-Autopilot #11726: PX4 jmavsim run issue in windows eclipse IDE

sanncal, 2019-03-27:
I am trying to run px4 firmware with jmavsim from within eclipse in Windows and after the GUI opens it hangs. However, if I run the same one from Cygwin bash command, it is working fine.
Can you please help me solve the issue.

Below is the console output in eclipse. If you look at the last few lines instead of getting the PX4> prompt to issue takeoff commands it is calling shutdown and then hangs until I close the GUI direct by clicking the close('x') button.

16:39:32 **** Build of configuration Default for project Firmware ****
make px4_sitl jmavsim 
[0/1] cd /cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware/build/px4_sitl_default/tmp && /cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware/Tools/sitl_run.sh /cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware/build/px4_sitl_default/bin/px4.exe none jmavsim none /cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware /cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware/build/px4_sitl_default/bin/px4.exe
debugger: none
program: jmavsim
model: none
src_path: /cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware
build_path: /cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware/build/px4_sitl_default/bin/px4.exe" "/cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware"/test_data
Buildfile: D:\GTO-ResearchEngineering\TALARIA-TuDLF\PX4\Project\Firmware\Tools\jMAVSim\build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 1 second
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

Init MAVLink
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Creating symlink /cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware/ROMFS/px4fmu_common -> /cygdrive/d/GTO-ResearchEngineering/TALARIA-TuDLF/PX4/Project/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF commencing GPS fusion
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-03-27/11_11_33.ulg
INFO  [px4] Startup script returned successfully
pxh> [2K
pxh> Shutting down
Shutting down...

16:41:54 Build Finished. 0 errors, 0 warnings. (took 2m:22s.275ms)

MaEtUgR, 2019-03-27:
I switched from eclipse to Visual Studio Code when it came out and one of the numerous reasons was that eclipse at least without additional plugins has a non-interactive shell to build stuff and my first guess is that this is your problem. When you start the simulation with e.g. jmavsim it runs the px4 binary in the build shell and you get into the interactive pxh shell. My guess is now that because elipse is not supporting any input on the shell it sends a SIGINT as soon as the process waits for input and that why the simulation shuts down again immediately. So you need to find a way to get the build shell interactive in eclipse or start the simulation in another way...

hkuyam008, 2019-03-29:
I am trying to do the exact same thing - "I am trying to run px4 firmware with jmavsim from within eclipse(C/C++ edition) in Windows 8"  (make px4_sitl jmavsim)

But for me the outcome is a bit different. The JMAVSIM simulator GUI opens and then the PX4 "auto initiates" a shutdown command (as evidenced by the console logs) and the GUI disappears/closes.

My concern is how is eclipse able to send the shutdown command, if eclipse does not support any input on the shell?

I am noticing the same behavior - "jMAVSIM GUI starts up and then shuts down gracefully, the console log shows shutdown command being issued" on Ubuntu LTS 14.04 as well.
In this case I am issuing the "make px4_sitl jmavsim" command from the terminal directly, NOT from within any IDE.

Is there some build config tweaking that we need to do to prevent this "auto shutdown of jmavsim"?

MaEtUgR, 2019-03-29:
So if I read correctly you're talking about two cases now:
1. You run the build command `make px4_sitl jmavsim` on Windows using the latest Cygwin Toolchain by configured make target in the GUI of the eclipse IDE which runs with the PATH set up by the Cygwin Toolchain batch script. Then eclipse wants to get that build through on its non-interactive shell and if I remember correctly it closes all interactive shells that wait for input along the way probably by sending the interrupt signal (SIGINT) to the shell which shuts down simulation and after that also closes the jMAVsim window. And because you wrote that it works in the interactive shell I was assuming that is the problem.
2. You run the build command `make px4_sitl jmavsim` on a completely separate Ubuntu 14.04 setup with a different clone and see the same shutdown result. There I have no idea what happens. And it means you are not able to run jMAVsim simulation in any way on that setup? You might want to compare to older PX4 versions and see if that changes anything. E.g. latest two stable releases 1.8.2 and 1.7.3 https://github.com/PX4/Firmware/releases.

hkuyam008, 2019-03-29:
Yes I was talking about 2 cases

1. You are correct in your understanding of my issue. 
    However you said "And because you wrote that it works in the interactive shell I was assuming that is 
    the problem" - I think that statement is still correct for the ticket creator. I am trying to do the same 
    thing as the ticket creator using the same tools and environment setup but getting a slightly different 
    outcome. 
    For him the GUI hangs (if I understood correctly), for me it simply shuts down/ closes.

2. You are 100% correct here - I think your suggestion to run a different release and see what happens is 
    great, I will try that out and post the results in this thread.

hkuyam008, 2019-04-01:
Update on this issue -

1. I ran "java px4_sitl jmavsim" with master branch (latest build) of Firmware on my Ubuntu laptop.
Observation - jmavsim GUI opens and closes, as reported before. (PX4 console output screenshot attached). 

2. Ran "make posix jmavsim" by checking out tag v1.8.2 to local branch on same Ubuntu laptop.
Observation - jmavsim GUI opens and closes as before, but PX4 console prints some additional error msg - "ERROR [urob] parameter_update sub failed" (screenshot attached)

Note - 
1. My laptop has Ubuntu LTS 14.04 installed
2. I am yet to test v1.8.2 for jmavsim on my Windows machine - wiil do it asap and share my findings here.

hkuyam008, 2019-04-01:
Just out of curiosity tried above "make posix jmavsim" with version as far back as v1.7.0. Interesting observation - this time the jmavsim GUI opens up and the px4 console / terminal hangs after printing the following 

"[1/1] cd /home/cloudset/PX4/src/Firmware/build/posix_sitl_default/tmp && /home/cloudset/PX4/src/Firmwa... jmavsim none /home/cloudset/PX4/src/Firmware /home/cloudset/PX4/src/Firmware/build/posix_sitl_default"

hkuyam008, 2019-04-02:
Update - running "make px4_sitl jmavsim" / "make posix jmavsim" on Windows 7 Enterprise SP1 machine from inside eclipse IDE

Observation 1 - for master branch / latest release of Firmware - jmavsim GUI opens and shuts down. PX4 Console log output below

Observation 2 - for v1.8.2 release of Firmware - jmavsim GUI opens and shuts down (same behavior as above). PX4 Console log output below

hkuyam008, 2019-04-02:
Comparing the log output in the above screenshots for the Windows run with the log output when "make posix jmavsim" is run from within Cygwin bash console, it seems like when run from within an IDE, it is having difficulty opening & writing to log file (my wild guess based on comparing just the output logs)

Where the above log prints "pxh> Shutting down", when run from Cygwin bash window, the next log lines are - 
pxh> INFO  [logger] Start file log
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2019-04-02/03_58_54.ulg
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF commencing GPS fusion

hkuyam008, 2019-04-03:
FYI... I have also tried running eclipse the way you suggested here - https://github.com/PX4/Firmware/issues/10902.

Outcome remains the same - The Simulator UI opens up and shuts down.

MaEtUgR, 2019-04-05:
I never ran the simulator as a build target from within eclipse, I just had the make target `posix` in eclipse to compile it and be able to jump to compile errors as they come up and then a cygwin shell to run the simulation with `make posix jmavsim` interactively. But I can try once on my PC.

sanncal, 2019-04-12:
@MaEtUgR 
So how you debug the PX4 code with things like Step over, Step into etc.? Any other IDE?

MaEtUgR, 2019-04-23:
@sanncal I didn't spend much time to get proper debug stepping in the IDE working yet even though it would be useful. Here are my latest findings: https://github.com/PX4/windows-toolchain/issues/7 So it's possible to run gdb in the Cygwin toolchain and I would use VS code as the IDE since it's also the officially supported one for debugging on linux (see configuration in the repository here: https://github.com/PX4/Firmware/tree/c50c44cc582956aefd32e39a6c7a5210ba4e1585/.vscode).

I can add the necessary components like gdb to the next toolchain installer and we need to find configuration to make cmake and gdb of the toolchain play with VS code.

closed 2019-08-05 by stale[bot]