PX4/PX4-Autopilot #11736: Pixracer external compass not detected with PX4 (and IS detected with Ardupilot)

copterspace, 2019-03-28:
Pixracer R14 + Radiolink TS100 GPS module (compass QMC5883L, https://ru.aliexpress.com/item/TS100-M8N-gps/32845084683.html). 
QGC v3.4.4.
When I flash Ardupilot firmware via QGC - the board does detect external compass OK.
Then I flash this board with PX4 firmware - it does not detect external compass.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash pixracer with Ardupilot 
[image]
[image]

2. See external compass 
[image]
[image]

3. Flash Pixracer with PX4 flight stack
[image]
[image]

4. External compass not detected
[image]
[image]

**Expected behavior**
Detect this compass with PX4 firmware, same good as Ardupilot.

It seems to be the same problem described in https://github.com/PX4/Firmware/issues/7607. I followed the recommendation "use QGC", but it did not help.
It also seems to be reason of problem I described here - http://discuss.px4.io/t/ekf2-yaw-diverging/10190 - because I used only internal compass, but it works badly.

copterspace, 2019-03-28:
Actually, driver for QMC5883L is developed, but not yet available in stable release: https://github.com/PX4/Firmware/pull/11140
Another aspect is that it is removed from autostart (https://github.com/PX4/Firmware/pull/11140/commits/c16c3b113ec521046252e366d3c7a9f7b451578b), even if I download developers release in QGC - must start it manually before it can be calibrated:
`qmc5883 start`

I downloaded fresh PX4 repository, added qmc5883 to autostart (https://github.com/PX4/Firmware/pull/11140/commits/c16c3b113ec521046252e366d3c7a9f7b451578b) - now it is visible in QGC: 
[image]

Now will resume the flight tests...

copterspace, 2019-03-29:
Flight test passed very good.
https://logs.px4.io/plot_app?log=08b75f05-133b-41d5-b940-4ba07247f7d1
I think, you may close this after you add QMC5883L compass driver + autostart to official release.

bkueng, 2019-04-01:
@dagar shouldn't we enable autostart so that we get it tested?

copterspace, 2019-04-01:
yes, surely, imho, you should. 
Because if you do not enable autostart - the only way to test it is to start it manually - then it resets after FCU reboots. It is very inconvenient way on real flying drone (you may not have an access to FCU console in the field).
Another way - add autostart to the source code and recompile custom firmware file - which I did. But it may be not very clear way for other users.

dagar, 2019-04-02:
> @dagar shouldn't we enable autostart so that we get it tested?

Yes, if someone has tested it lets get it added to autostart.

closed 2019-04-15 by bkueng