PX4/PX4-Autopilot #11737: Inconsistent pre flight fails in SITL

julianoes, 2019-03-28:
On latest master (9a2617ad335fb5f1f3ac7cd5cf1f674439d58ee4) I'm flying missions in SITL Gazebo with the plane model and often times can't arm again after landing.

The preflight failure on arming is:
```
WARN  [commander] Preflight Fail: High Accelerometer Bias
```

However, the `estimator_status` seems fine at this moment, and I have not observed failures during the flight before.
```
pxh> listener estimator_status

TOPIC: estimator_status instance 0 #1
 estimator_status_s
	timestamp: 374420000  (0.004000 seconds ago)
	states: [0.6922, -0.0036, 0.0013, 0.7217, -0.0094, 0.0293, 0.0216, -11.3129, 298.1385, 0.0230, 0.0000, -0.0000, 0.0000, -0.0025, 0.0010, -0.0013, 0.2161, 0.0078, 0.4299, 0.0008, 0.0007, -0.0005, 0.2182, -0.2205]
	vibe: [0.0000, 0.0000, 0.0003]
	covariances: [0.0000, 0.0000, 0.0000, 0.0000, 0.0162, 0.0162, 0.0112, 0.0953, 0.0953, 0.0576, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000]
	control_mode_flags: 131607
	pos_horiz_accuracy: 0.4366
	pos_vert_accuracy: 0.2400
	mag_test_ratio: 0.0402
	vel_test_ratio: 0.0147
	pos_test_ratio: 0.0026
	hgt_test_ratio: 0.0114
	tas_test_ratio: 0.3714
	hagl_test_ratio: 0.0000
	beta_test_ratio: 0.0041
	time_slip: 0.0000
	gps_check_fail_flags: 0
	filter_fault_flags: 0
	innovation_check_flags: 0
	solution_status_flags: 831
	n_states: 24
	pre_flt_fail: False
	health_flags: 0
	timeout_flags: 0
```

The situation doesn't get resolved unless I restart SITL. I remember having seen similar weird issues out in the field with a multicopter.

julianoes, 2019-03-28:
This is the check that fails:
```
checking index: 13: status: -0.002560, test_limit: 0.001730, test_uncertainty: 0.000678
```

https://github.com/PX4/Firmware/blob/75f1ad36d1210560f00da8f4fa3bddbb3da9da16/src/modules/commander/PreflightCheck.cpp#L585

julianoes, 2019-03-28:
What confuses me is that there are these magic check numbers in commander when really the ekf2 itself should decide if it's happy about its estimate or not. I would assume that the commander only needs to check the estimator flags from `estimator_status`.

@priseborough I would appreciate your input, thanks.

julianoes, 2019-03-28:
Again here, a couple of flights worked, until finally the estimate was too big:

```
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF commencing GPS fusion
[Wrn] [Publisher.cc:141] Queue limit reached for topic /wind, deleting message. This warning is printed only once.
checking index: 13: status: 0.000070, test_limit: 0.001730, test_uncertainty: 0.001743
checking index: 14: status: 0.000033, test_limit: 0.001730, test_uncertainty: 0.001743
checking index: 15: status: -0.000625, test_limit: 0.001730, test_uncertainty: 0.000671
INFO  [commander] Takeoff detected
INFO  [commander] Landing detected
INFO  [logger] closed logfile, bytes written: 9378325

pxh> checking index: 13: status: -0.002061, test_limit: 0.001730, test_uncertainty: 0.000674
checking index: 14: status: 0.001473, test_limit: 0.001730, test_uncertainty: 0.000674
checking index: 15: status: -0.001262, test_limit: 0.001730, test_uncertainty: 0.000671
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-03-28/10_10_09.ulg
INFO  [commander] Takeoff detected
INFO  [commander] Landing detected
INFO  [logger] closed logfile, bytes written: 6887234

pxh> checking index: 13: status: -0.001840, test_limit: 0.001730, test_uncertainty: 0.000678
checking index: 14: status: 0.000738, test_limit: 0.001730, test_uncertainty: 0.000678
checking index: 15: status: -0.001412, test_limit: 0.001730, test_uncertainty: 0.000671
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-03-28/10_12_47.ulg
INFO  [commander] Takeoff detected
pxh> INFO  [commander] Landing detected
pxh> INFO  [logger] closed logfile, bytes written: 6949307

checking index: 13: status: -0.001866, test_limit: 0.001730, test_uncertainty: 0.000671
checking index: 14: status: 0.002135, test_limit: 0.001730, test_uncertainty: 0.000671
checking index: 15: status: -0.000692, test_limit: 0.001730, test_uncertainty: 0.000671
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-03-28/10_13_36.ulg
INFO  [commander] Takeoff detected
INFO  [commander] Landing detected
INFO  [logger] closed logfile, bytes written: 6894766

checking index: 13: status: -0.002014, test_limit: 0.001730, test_uncertainty: 0.000671
checking index: 14: status: 0.002116, test_limit: 0.001730, test_uncertainty: 0.000671
checking index: 15: status: -0.001168, test_limit: 0.001730, test_uncertainty: 0.000671
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-03-28/10_14_24.ulg
INFO  [commander] Takeoff detected
INFO  [commander] Landing detected
INFO  [logger] closed logfile, bytes written: 6887258

checking index: 13: status: -0.002488, test_limit: 0.001730, test_uncertainty: 0.000673
WARN  [commander] Preflight Fail: High Accelerometer Bias
checking index: 13: status: -0.002488, test_limit: 0.001730, test_uncertainty: 0.000673
WARN  [commander] Preflight Fail: High Accelerometer Bias
checking index: 13: status: -0.002488, test_limit: 0.001730, test_uncertainty: 0.000674
WARN  [commander] Preflight Fail: High Accelerometer Bias

```

priseborough, 2019-04-03:
The problem is that what constitutes 'normal'  value are contextual, eg preflight waiting for takeoff vs in flight,  a static vs moving rover, etc. High bias values are not a problem for the EKF, but often indicate issues with sensor calibration that can cause problems later in flight. 

We have tried to strike a balance between false negative and positive rates. The fundamental issue with accel bias is that until the vehicle starts to e and rotate, the bias for axes perpendicular to the gravity vector are poorly observable.

Please provide a log that shows this issue with pre-arm logging enabled.

julianoes, 2019-04-24:
Here is another log, this time I wanted to takeoff with the deltaquad and could not arm:
https://logs.px4.io/plot_app?log=cd8d6902-ca8e-4525-aea3-934e15608f0c

julianoes, 2020-03-01:
Closing for now. I'll bring re-open this if it happens again.

closed 2020-03-01 by julianoes