PX4/PX4-Autopilot #11739: Spurious MANUAL CONTROL LOST on Pixracer

julianoes, 2019-03-28:
Every few seconds, I see the following warning on a Pixracer with only USB and debug console connected:

```
WARN  [commander] MANUAL CONTROL LOST (at t=17692ms)
```

So far I've found that `manual_control_setpoint` is published  because `input_rc` has been published. I have however not found who is publishing `input_rc`.

LorenzMeier, 2019-03-28:
Is your GCS sending joystick commands? Do you have on-Screen or real joystick enabled?

julianoes, 2019-03-28:
> Is your GCS sending joystick commands? Do you have on-Screen or real joystick enabled?

No, that was my initial suspect as well.

According to the `rc_input` struct, the input source is PX4FMU SBUS:

```
listener input_rc

TOPIC: input_rc instance 0 #1
 input_rc_s
        timestamp: 152511328  (10.958436 seconds ago)
        timestamp_last_signal: 152511328
        channel_count: 18
        rssi: 255
        rc_lost_frame_count: 1
        rc_total_frame_count: 0
        rc_ppm_frame_length: 0
        values: [874, 1036, 1274, 1394, 874, 884, 874, 878, 884, 1204, 1004, 1208, 2143, 2135, 2033, 1842, 998, 998]
        rc_failsafe: False
        rc_lost: False
        input_source: 9
```
It looks like sbus is parsing something.

LorenzMeier, 2019-03-28:
What is connected to that port? Anything? Nothing? Where did you purchase the bird from?

julianoes, 2019-03-28:
It's a Pixracer R15 in purple, presumably from mRo. Nothing is connected.

julianoes, 2019-03-28:
It feels like it's floating and reading garbage.

julianoes, 2019-03-28:
This is what it looks like:
```
new bytes: 1: 0x00 
new bytes: 1: 0x02 
new bytes: 2: 0x00 00 
new bytes: 4: 0x00 01 80 48 
new bytes: 8: 0x90 d0 06 41 00 00 18 a1 
new bytes: 2: 0xa2 1d 
new bytes: 6: 0x00 00 11 09 cc 7e 
new bytes: 1: 0xfe 
new bytes: 8: 0xf4 7b 53 50 00 06 3c 62 
new bytes: 8: 0x01 91 40 00 81 00 20 18 
new bytes: 6: 0x02 b8 00 ae 3f f2 
new bytes: 3: 0xff fe f0 
new bytes: 1: 0x00 
new bytes: 1: 0x00 
new bytes: 4: 0x00 00 10 00 
new bytes: 5: 0x81 10 00 20 00 
new bytes: 9: 0x00 0c 0e 00 18 26 bb f0 fb 
new bytes: 16: 0xff f8 e1 7e bf ff 7e e8 d0 ed 93 76 8f 22 20 01 
new bytes: 7: 0x1c 07 40 40 00 00 00 
new bytes: 6: 0x00 00 60 00 08 38 
new bytes: 5: 0x0c 03 de ff ff 
new bytes: 2: 0x00 02 
new bytes: 10: 0x00 a2 3d 20 20 70 ee fd f0 ff 
new bytes: 13: 0xf0 fa f0 56 9e f0 ff bc ff 86 03 00 40 
new bytes: 7: 0x07 2c e0 00 00 00 00 
new bytes: 6: 0x00 00 40 08 0f e0 
new bytes: 5: 0x00 71 04 5a f0 
new bytes: 21: 0xfe f9 f0 fb d0 da ce d4 5e b5 7e 00 20 66 bc 00 81 00 d3 7b c1 
we parsed from SBUS
received RC in
publishing sp_man
new bytes: 1: 0xf4 
copy sp_man
WARN  [commander] MANUAL CONTROL LOST (at t=11511ms)
```

I tried another identical Pixracer and the problem does not happen, however, it consistently reads one 0x00 byte about every 2 seconds. Presumably that's a different problem.

Closing this as it's apparently just faulty hardware.

closed 2019-03-28 by julianoes