PX4/PX4-Autopilot #11740: Large  fluctuation with MS5611 Airspeed sensor

jstrebel, 2019-03-28:
Hi,
I use a holybro speed sensor MS5611 and notice very large fluctuations of the airspeed measurements in idle mode (closed room). I have recalibrated this sensor several times. The result is always the same. Unfortunately there is no second sensor available for comparison. A comparison with a SDP3 from drotek shows a better behaviour. But I also see artefacts in this log.
Question: Is the sensor defective or were there changes in 1.9beta that cause this? I see that the MS5611 was added after the 1.8.2 release.
Software: Px4 version 1.9beta
Hardware: Holybro Pixhawk 4
https://review.px4.io/plot_app?log=5a97e051-a43a-455d-bbf9-1dfba47c2bc9
https://review.px4.io/plot_app?log=b072ef9c-2f10-4b69-a17f-5b6c906364e4

jstrebel, 2019-03-29:
Negative Airspeed, never saw this with SDP3.
[image]

LorenzMeier, 2019-03-29:
These sensors work completely differently - the MS5611 measures differential pressure while the SDP3 measures the cooling of a CMOS element by passing air molecules (the pitot in the case of the SDP3 is "leaking" for this purpose, which also explains why it needs a matched / calibrated pitot).

That also means they have *completely* different noise characteristics. What you are seeing is the effect of the square root in the airspeed formula:
https://en.wikipedia.org/wiki/True_airspeed#Calculating_true_airspeed

It means that with increasing speed, your relative noise level and steady state error goes down, as the same pressure offset and noise (absolute) translates to an ever-decreasing velocity error. And for the same reason negative airspeed makes sense: The sensor is oscillating around zero (the calibrated value). The filter will reject these measurements though.

closed 2019-03-29 by LorenzMeier

jstrebel, 2019-03-30:
@LorenzMeier, Thank you Lorenz for the detailed explanation. The reason I raised this was on the fact that I did some trials with a bungee start of a Phantom Wing, which was not successful. To reduce the damage I did tests just "shooting" the plane in the air without the motor. The launch trigger was detected, but the Wing-wing (px4)  did not try to give any pitch in the starting phase. Thomas Stastney looked at the log and he raised the suspicion, the airspeed sensor could be the culprit. So I looked at this sensor and compared it with SDP3.

ryanjAA, 2019-04-01:
We see negative airspeed all the time on the sdp3 units and the interesting thing is that while Lorenz is completely correct about their low speed performance, from a practical standpoint one can tolerate more of an error at higher speeds since in fixed wing we won’t fall out of the sky if going x percentage faster or slower when at cruise speed but at min or stall speed, we need to be on the mark. That’s the benefit we’ve seen with the sdp3 units but they are a pain to work with initially.

ryanjAA, 2019-04-01:
I am unsure about the matched/calibrated pitot though. So long as the tube length is entered and kept at 1.5mm I.D. and the pitots are kept the same, when next to each other, we get the same readings when testing but I gather that’s what @lorenz meant about calibrated pitots (correct me if I’m wrong please so I can ask for some different types of testing to be done).

LorenzMeier, 2019-04-01:
@ryanjAA In theory it also depends on the type of pitot tube, yes, the ID should be 1.5mm and you need to set up CAL_AIR_TUBELEN as precisely as possible.

jstrebel, 2019-04-01:
I would be willing to do a test without a pitot. Would this mean just take a tube which a entry hole of (lets say 1mm diameter? and do a calibration.

ryanjAA, 2019-04-02:
Thanks @lorenzmeier makes sense. @jstrebel Lorenz can better answer that since there is some black box magic (ie a pressure drop model implemented for the sensor). I believe this is more of a flow sensor, not differential pressure sensor but I’m not sure if the model works without the difference between the readings from the static and dynamic ports from the pitot. Just get a steady wind source (fan or otherwise) and do some basic bench testing first (although I’m sure Lorenz, reading through the model if it’s available or whomever wrote the model may know quite quickly. Compare the non pitot with readings from with a pitot or from another sensor. Would be interested to know.

dagar, 2019-04-02:
Minor correction for anyone that stumbles across this later. The Holybro sensor is an MS4525 differential pressure sensor, not an MS5611 (a barometric pressure sensor).