PX4/PX4-Autopilot #11741: PX4 "Recject Auto mission" and "Reject auto takeoff"

marcel7797, 2019-03-28:
Hi, I'm quite new to this and it seems that this error is beyond my knowledge so I'm writing it down to see if someone has the same issue.
I have a PX4 mounted on a fixed wing with GPS and telemetry (the basic configuration).

The situation is that when I load a simple mission ( takeoff, waypoint and land) when I activate mission mode though a switch in mi RC controller the message "Reject Auto mission" appears and if the mission is executed with Qgroundcontrol the message is "unable to start mission: vehicle not ready". I have GPS with 3D lock when the error appears and the log file don't present any anomalies or unexpected readings. 
I've tried reinstalling the firmware (PX4 1.8.2) and used another PX4 and the error keeps showing. With auto take off the error is the same.
Manual mode and stabilized works as expected.

Maybe I'm doing something wrong and I'm not realizing it...
Thanks!

LorenzMeier, 2019-03-28:
An error report without a log file is impossible to manage - please provide a flight log for this. Thanks!

marcel7797, 2019-03-28:
I've ran this test now and when I try to pas from stabilized mode to mission or takeoff the controller shows the mentioned error and and changes to altitude mode. 
https://logs.px4.io/plot_app?log=54649a2b-1881-466e-bc9c-e112e9b98a2e

LorenzMeier, 2019-03-29:
You don't have a suitable GPS accuracy. The uncertainty is 20 meters. Please check the installation of your GPS unit, make sure it has good sky view and is not too close to other electronics. You also need to be in an open field (not indoors, not between tall buildings).

[image]

closed 2019-03-29 by LorenzMeier

marcel7797, 2019-03-29:
Okay, that's something to work with! Thank a lot!