PX4/PX4-Autopilot #11743: PX4 Navio PWM

alexisutc, 2019-03-29:
Hello world,

I use following config:

- RPi3 + Navio2
- PX4-emlid-navio2 firmware from [here](https://dev.px4.io/en/setup/building_px4.html) 

I got two problems here:

**Little A**
The px4 firmware was installed just fine.
But: when I try to use the PWM output with this code for example :
`pwm test -c 13 -p 1200`
I got following error message
`ERROR [pwm] can't open /dev/pwm_output0`
I guess this comes from the hardware configuration : I’m using the navio and not a pixhawk, that's why i don't have the pwm_output0 driver file...
in `/dev/` there are no such file. But the is a driver in following folder:
`/sys/class/pwm/pwmchip0`

**Little B**
I want to have a mavlink communication between the PC and the RPi.
When i write following code in the px4 shell i succeed to connect to the GCS
`mavlink start -u 14551 -o 14550 -t XXX.XXX.XXX.XXX'
X represents the real IP numbers of the GCS
So far so good.
But I want to do the same with ROS
So basically I start the mavros node with following command :
`rosrun mavros mavros_node \
_fcu_url:=udp://:14650@ \
_gcs_url:=udp://:14551@XXX.XXX.XXX.XXX:14550`

I also tried like this

`rosrun mavros mavros_node \
_fcu_url:=udp://YYY.YYY.YYY.YYY:14650@ \
_gcs_url:=udp://YYY.YYY.YYY.YYY:14551@XXX.XXX.XXX.XXX:14550`

with Y the IP of the FCU

but without success.

Any idea?

closed 2019-07-11 by stale[bot]

julianoes, 2019-07-15:
@alexisutc which instructions did you follow for the Navio2? Do they offer some kind of help or support?

alexisutc, 2019-07-16:
hello,

i followed some tutorials specially on the emlid webside.
Actually i solved my problem in a different way. I used ardupilot and not px4. I changed the source code of it and made an own node to write on the pwm.
Working just fine now.

bye

alexis