PX4/PX4-Autopilot #11747: EKF2 Airspeed estimate wrong in high wind (again)

ryanjAA, 2019-03-31:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

julianoes, 2019-07-15:
@RomanBapst something for you

moreba1, 2020-03-01:
what is method of calculating wind speed in px4 ? 
i think there is a simple way for calculating by subtract true airspeed and groundspeed vectors .

sfuhrer, 2020-03-03:
> what is method of calculating wind speed in px4 ?
> i think there is a simple way for calculating by subtract true airspeed and groundspeed vectors .
> 
> @sfuhrer
> @dagar

Yes groundspeed - groundspeed is used, as well as sideslip fusing (you know there is wind from the side if the vehicle is moving lateral to it's heading). 

See https://github.com/PX4/ecl/blob/master/airdata/WindEstimator.cpp#L1 for the code. It's a standalone, small EKF only for wind estimation.

bresch, 2022-08-15:
The airspeed scale factor was probably wrong. `main` does now estimate it as well

[image]

closed 2022-08-15 by bresch

ryanjAA, 2022-08-15:
From the above you would just increase the scale by  0.1 as a starting point?

bresch, 2022-08-15:
@ryanjAA set `ASPD_SCALE_1` to 0.9. In recent version of the code (probably already from 1.12), the scale is estimated by a separate estimator (not EKF2) and then used to re-scale the airspeed data that is used by other modules (EKF2, TECS, ...) so it should already be better even without setting the `ASPD_SCALE_1` parameter (but it's better if it's set to the correct value).
You can also look at a recent log to see to which value the estimator converged (see https://github.com/PX4/PX4-Autopilot/blob/d7a962b4269d3ca3d2dcae44da7a37177af1d8cd/msg/airspeed_wind.msg#L13)