PX4/PX4-Autopilot #11750: Firmware broken on Crazyflie 2.0 hardware

dcabecinhas, 2019-03-31:
**Describe the bug**

The crazyflie 2.0 platform, using the latest Firmware, does not communicate with QGC through USB nor through the PA radio.

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest master or 1.9.0beta Firmware in Crazyflie 2.0 hardware
2. Connect to QGC using USB or the PA radio (with cfbridge)
3. "Waiting for connection" is displayed on QGC indefinitely

**Expected behavior**
QGC should display vehicle status and parameters

**Log Files**

Using cflib 1.6.0 and cfbridge from https://github.com/dennisss/cfbridge

Output of cfbridge with latest firmware (or 1.9.0beta) [nothing shows up on QGC]

	Scanning interfaces for Crazyflies...
	DEBUG:cflib.crtp:Scanning: <class 'cflib.crtp.radiodriver.RadioDriver'>
	INFO:cflib.crtp.radiodriver:v0.53 dongle with serial N/A found
	DEBUG:cflib.crtp:Scanning: <class 'cflib.crtp.serialdriver.SerialDriver'>
	DEBUG:cflib.crtp:Scanning: <class 'cflib.crtp.udpdriver.UdpDriver'>
	DEBUG:cflib.crtp:Scanning: <class 'cflib.crtp.usbdriver.UsbDriver'>
	INFO:cflib.drivers.cfusb:Looking for devices....
	Crazyflies found:
	radio://0/80/2M
	DEBUG:cflib.crazyflie:Adding callback on port [6] to [<bound method Localization._incoming of <cflib.crazyflie.localization.Localization object at 0x10b8c0550>>]
	DEBUG:cflib.crazyflie:Adding callback on port [5] to [<bound method Log._new_packet_cb of <cflib.crazyflie.log.Log object at 0x10b8c0668>>]
	DEBUG:cflib.crazyflie:Adding callback on port [0] to [<bound method Console.incoming of <cflib.crazyflie.console.Console object at 0x10b8c0710>>]
	DEBUG:cflib.crazyflie:Adding callback on port [2] to [<bound method _ParamUpdater._new_packet_cb of <_ParamUpdater(Thread-2, initial daemon)>>]
	DEBUG:cflib.crazyflie:Adding callback on port [4] to [<bound method Memory._new_packet_cb of <cflib.crazyflie.mem.Memory object at 0x10b8c09b0>>]
	Connecting to radio://0/80/2M
	INFO:cflib.crazyflie:Callback->Connection initialized[radio://0/80/2M]
	INFO:cflib.crazyflie:We are connected[radio://0/80/2M], request connection setup
	DEBUG:cflib.crazyflie:Sending packet and expecting the (93, 5) pattern back

	waiting to receive message
	INFO:cflib.crazyflie:Resending for pattern (93, 5)
	DEBUG:cflib.crazyflie:We want to resend and the pattern is there
	INFO:cflib.crazyflie:Resending for pattern (93, 5)
	DEBUG:cflib.crazyflie:We want to resend and the pattern is there
	INFO:cflib.crazyflie:Resending for pattern (93, 5)
	DEBUG:cflib.crazyflie:We want to resend and the pattern is there
	INFO:cflib.crazyflie:Resending for pattern (93, 5)
	DEBUG:cflib.crazyflie:We want to resend and the pattern is there
	INFO:cflib.crazyflie:Callback->Connected to [radio://0/80/2M]
	DEBUG:cflib.crazyflie:Looking for pattern match on (93, 5) vs (255, 1, 35)
	DEBUG:cflib.crazyflie:Looking for pattern match on (93, 5) vs (255, 1, 35)
	DEBUG:cflib.crazyflie:Looking for pattern match on (93, 5) vs (93, 5, 103, 0)
	DEBUG:cflib.crazyflie:Found new longest match (93, 5)
	DEBUG:cflib.crazyflie.log:Logging reset, continue with TOC download
	DEBUG:cflib.crazyflie.toc:[5]: Start fetching...
	DEBUG:cflib.crazyflie:Adding callback on port [5] to [<bound method TocFetcher._new_packet_cb of <cflib.crazyflie.toc.TocFetcher object at 0x10b8dc208>>]
	DEBUG:cflib.crazyflie.toc:[5]: Got TOC CRC, 0 items and crc=0x00000000
	DEBUG:cflib.crazyflie.toc:Requesting index 0 on port 5
	DEBUG:cflib.crazyflie.toc:Added element [0]
	WARNING:cflib.crazyflie.toccache:Could not save cache, no writable directory
	DEBUG:cflib.crazyflie:Removing callback on port [5] to [<bound method TocFetcher._new_packet_cb of <cflib.crazyflie.toc.TocFetcher object at 0x10b8dc208>>]
	DEBUG:cflib.crazyflie.toc:[5]: Done!
	INFO:cflib.crazyflie:Log TOC finished updating
	INFO:cflib.crazyflie.mem:Requesting number of memories
	INFO:cflib.crazyflie.mem:0 memories found
	INFO:cflib.crazyflie:Memories finished updating
	DEBUG:cflib.crazyflie.toc:[2]: Start fetching...
	DEBUG:cflib.crazyflie:Adding callback on port [2] to [<bound method TocFetcher._new_packet_cb of <cflib.crazyflie.toc.TocFetcher object at 0x10b8dc208>>]
	DEBUG:cflib.crazyflie.toc:[2]: Got TOC CRC, 0 items and crc=0x00000000
	DEBUG:cflib.crazyflie.toc:Requesting index 0 on port 2
	DEBUG:cflib.crazyflie.toc:Added element [0]
[... 96 lines cut ...]
	INFO:cflib.crazyflie:Callback->Connection setup finished [radio://0/80/2M]
	Connected to radio://0/80/2M
	DEBUG:cflib.crazyflie.param:Requesting request to update param [0]
	DEBUG:cflib.crazyflie:Sending packet and expecting the (45, 0) pattern back
	DEBUG:cflib.crazyflie:Looking for pattern match on (45, 0) vs (255, 1, 45)
	DEBUG:cflib.crazyflie:Looking for pattern match on (45, 0) vs (255, 1, 45)
	DEBUG:cflib.crazyflie:Looking for pattern match on (45, 0) vs (255, 1, 44)
	DEBUG:cflib.crazyflie:Looking for pattern match on (45, 0) vs (45, 0, 0)
	DEBUG:cflib.crazyflie:Found new longest match (45, 0)
	DEBUG:cflib.crazyflie.param:Updated parameter [.]
	received 21 bytes from ('127.0.0.1', 14550)

	waiting to receive message
	received 13 bytes from ('127.0.0.1', 14550)

	waiting to receive message
	received 43 bytes from ('127.0.0.1', 14550)

	waiting to receive message
	received 19 bytes from ('127.0.0.1', 14550)

	waiting to receive message
	received 19 bytes from ('127.0.0.1', 14550)

	(...)

dcabecinhas, 2019-03-31:
I was doing a bisection on the Firmware commits and the connection troubles seem to have started with 48f9d3f3d92e2c8dcd7d364665909edfe741a89f

dagar, 2019-07-14:
Is PX4 on the Crazyflie 2 actively used by anyone? Let's decide if we actually want to keep it supported and I'll get one on the test rack.

julianoes, 2019-07-15:
@bkueng do you have an opinion here?

bkueng, 2019-07-16:
I'm torn. On one hand we never had good flight performance (mostly due to its sensory design, using an IMU via I2C), on the other hand it would be an interesting platform.

julianoes, 2019-08-29:
@spheroidal that's cool. And could anyone help us to support and maintain PX4 on crazyflie 2.0? And for instance help with this issue here?

spheroidal, 2019-08-30:
@julianoes I've been trying to debug this issue. I found that https://github.com/PX4/Firmware/commit/3859bbb3b0abdfa1314bb070e4d4aa4aec5d883b#diff-2dc55367f11ba4d01e3ba810ed3fe35d is the last commit that works on the crazyflie 2.0. After the pull request https://github.com/PX4/Firmware/pull/11256/commits, which updates the Nuttx, the codes still compile normally but didn't work after it was flashed to the crazyflie 2.0. Would you mind taking a look at what could be the problem in that pull request? I am not familiar with Nuttx and get stuck in debugging this... Thanks a lot!

julianoes, 2019-09-03:
I don't have the time or hardware, I'm afraid.

Do you have a serial console where you would see startup output to get an idea where it fails?

spheroidal, 2019-09-03:
Sorry, I don't have a serial console now. I could try that when I have time.

hamishwillee, 2019-12-02:
Ping

dagar, 2020-07-27:
Should be fixed in https://github.com/PX4/Firmware/pull/15421.

bresch, 2021-04-30:
This is fixed

closed 2021-04-30 by bresch