PX4/PX4-Autopilot #11751: -Werror=cast-align problem on beaglebone_blue_cross

teknokita, 2019-03-31:
I just tried build px4 master branch for beaglebone_blue and get this error:
-Werror=cast-align

**To Reproduce**
make beaglebone_blue_cross 
using tool and lib:
-gcc-linaro-7.4.1-2019.02-x86_64_arm-linux-gnueabihf
-librobotcontrol_1.0.4

get error:
librobotcontrol -> include/rc/mavlink/common/../protocol.h:280:49: error: cast from ‘const char*’ to ‘const uint16_t* {aka const short unsigned int*}’ increases required alignment of target type [-Werror=cast-align]

fix with -Wno-cast-align
file: Firmware/cmake/px4_add_common_flags.cmake

@@ -66,7 +66,7 @@ function(px4_add_common_flags)
		-Werror

		-Warray-bounds
-		-Wcast-align
+		-Wno-cast-align
		-Wdisabled-optimization
		-Wdouble-promotion
		-Wfatal-errors

dagar, 2019-04-02:
We need to get Beaglebone into the CI system.

@teknokita you can fix this by ignoring the warning (-Wno-cast-align) on a per module/driver basis. 

Example: https://github.com/PX4/Firmware/blob/0b548fa6849e7a608b4c120fb66a760e51806056/src/drivers/distance_sensor/sf1xx/CMakeLists.txt#L37

julianoes, 2020-01-30:
@dagar we do now have a beaglebone in CI, right?

dagar, 2020-01-30:
> @dagar we do now have a beaglebone in CI, right?

Yes.

closed 2020-01-30 by dagar