PX4/PX4-Autopilot #11757: sitl_gazebo build broken on macOS

julianoes, 2019-04-01:
On a fresh Mac, I'm trying to set up SITL with Gazebo.

I've done:
```
brew tap PX4/px4
brew install px4-dev
brew cask install xquartz
brew install px4-sim
```

And the build fails with:
```
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/src/geo_mag_declination.cpp:48:
/Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/common.h:68:10: warning: expression result unused [-Wunused-value]
    for( e_model; e_model; e_model=e_model->NextSiblingElement("model") )
         ^~~~~~~
1 warning generated.
[36/78] Building CXX object CMakeFiles/gazebo_multirotor_base_plugin.dir/src/gazebo_multirotor_base_plugin.cpp.o
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/src/gazebo_multirotor_base_plugin.cpp:21:
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/gazebo_multirotor_base_plugin.h:29:
/Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/common.h:68:10: warning: expression result unused [-Wunused-value]
    for( e_model; e_model; e_model=e_model->NextSiblingElement("model") )
         ^~~~~~~
1 warning generated.
[37/78] Building CXX object CMakeFiles/gazebo_vision_plugin.dir/src/gazebo_vision_plugin.cpp.o
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/src/gazebo_vision_plugin.cpp:41:
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/gazebo_vision_plugin.h:46:
/Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/common.h:68:10: warning: expression result unused [-Wunused-value]
    for( e_model; e_model; e_model=e_model->NextSiblingElement("model") )
         ^~~~~~~
1 warning generated.
[38/78] Building CXX object CMakeFiles/gazebo_motor_model.dir/src/gazebo_motor_model.cpp.o
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/src/gazebo_motor_model.cpp:22:
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/gazebo_motor_model.h:38:
/Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/common.h:68:10: warning: expression result unused [-Wunused-value]
    for( e_model; e_model; e_model=e_model->NextSiblingElement("model") )
         ^~~~~~~
1 warning generated.
[39/78] Building CXX object CMakeFiles/gazebo_irlock_plugin.dir/src/gazebo_irlock_plugin.cpp.o
[40/78] Building CXX object CMakeFiles/gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:22:
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/gazebo_mavlink_interface.h:57:
/Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/common.h:68:10: warning: expression result unused [-Wunused-value]
    for( e_model; e_model; e_model=e_model->NextSiblingElement("model") )
         ^~~~~~~
/Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/common.h:68:10: warning: expression result unused [-Wunused-value]
    for( e_model; e_model; e_model=e_model->NextSiblingElement("model") )
         ^~~~~~~
/Users/julianoes/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:339:3: note: in instantiation of function template specialization 'gazebo::model_param<int>' requested here
  model_param(worldName, model_->GetName(), "mavlink_udp_port", mavlink_udp_port_);
  ^
2 warnings generated.
[41/78] Building CXX object CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o
[42/78] Building CXX object CMakeFiles/gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o
[43/78] Building CXX object CMakeFiles/gazebo_uuv_plugin.dir/src/gazebo_uuv_plugin.cpp.o
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/src/gazebo_uuv_plugin.cpp:25:
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/gazebo_uuv_plugin.h:23:
/Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/common.h:68:10: warning: expression result unused [-Wunused-value]
    for( e_model; e_model; e_model=e_model->NextSiblingElement("model") )
         ^~~~~~~
1 warning generated.
[44/78] Building CXX object CMakeFiles/gazebo_imu_plugin.dir/src/gazebo_imu_plugin.cpp.o
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:21:
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/gazebo_imu_plugin.h:33:
/Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/common.h:68:10: warning: expression result unused [-Wunused-value]
    for( e_model; e_model; e_model=e_model->NextSiblingElement("model") )
         ^~~~~~~
1 warning generated.
[45/78] Building CXX object CMakeFiles/gazebo_sonar_plugin.dir/src/gazebo_sonar_plugin.cpp.o
[46/78] Linking CXX shared library libsensor_msgs.dylib
[... 244 lines cut ...]
1 warning generated.
[67/78] Building CXX object CMakeFiles/gazebo_controller_interface.dir/src/gazebo_controller_interface.cpp.o
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/src/gazebo_controller_interface.cpp:22:
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/gazebo_controller_interface.h:36:
/Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/common.h:68:10: warning: expression result unused [-Wunused-value]
    for( e_model; e_model; e_model=e_model->NextSiblingElement("model") )
         ^~~~~~~
1 warning generated.
[68/78] Building CXX object CMakeFiles/LiftDragPlugin.dir/src/liftdrag_plugin/liftdrag_plugin.cpp.o
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/src/liftdrag_plugin/liftdrag_plugin.cpp:21:
/Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/common.h:68:10: warning: expression result unused [-Wunused-value]
    for( e_model; e_model; e_model=e_model->NextSiblingElement("model") )
         ^~~~~~~
1 warning generated.
[69/78] Building CXX object CMakeFiles/gazebo_wind_plugin.dir/src/gazebo_wind_plugin.cpp.o
In file included from /Users/julianoes/src/Firmware/Tools/sitl_gazebo/src/gazebo_wind_plugin.cpp:23:
/Users/julianoes/src/Firmware/Tools/sitl_gazebo/include/common.h:68:10: warning: expression result unused [-Wunused-value]
    for( e_model; e_model; e_model=e_model->NextSiblingElement("model") )
         ^~~~~~~
1 warning generated.
[70/78] Building CXX object CMakeFiles/gazebo_opticalflow_plugin.dir/src/gazebo_opticalflow_plugin.cpp.o
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

julianoes, 2019-04-01:
Shouldn't this be covered by CI?

julianoes, 2019-04-01:
Related to: https://github.com/PX4/sitl_gazebo/issues/298

smnogar, 2019-04-02:
This looks like an opencv4 issue. I recommend sticking with opencv3 until its implemented in sitl_gazebo. 

* `brew install opencv@3` 
* `brew unlink opencv`
* `brew link opencv@3`

julianoes, 2019-04-02:
@smnogar actually the first error for opencv has been fixed here: https://github.com/PX4/sitl_gazebo/pull/299

julianoes, 2019-04-03:
Last fix coming up with https://github.com/PX4/sitl_gazebo/pull/311.

PR #11793 (2019-04-05): sitl_gazebo: update submodule
This should fix the macOS build.

Fixes #11757.
closed 2019-04-05 by dagar