PX4/PX4-Autopilot #11759: fix wing mission and altitude transtition fail

dannyfpv, 2019-04-01:
fantom wing pixhawk 1 
Vehicle arms, takes off and fly as expected, but once mission plan (automated) is triggered, fixed wing accelerates towards ground with the nose facing down.

pr and master some issues
https://review.px4.io/plot_app?log=d1c75dae-a1fe-4fc6-b6aa-20cb1f979aa8

dagar, 2019-04-06:
Could you share the mission you planned? Either the file or a screenshot of QGroundControl.

According to the log the waypoint altitude was about 66 meters and when you engaged mission mode you were at 228 meters.

[image]

dannyfpv, 2019-04-08:
[image]

It was flown at a high altitude for testing but the same issue can be seen while flying at low altitude.

dagar, 2019-04-08:
Could you go to the plan screen to show the altitudes?

dannyfpv, 2019-04-08:
[image]
plan screen

dannyfpv, 2019-04-08:
another log fron fixed wing
https://review.px4.io/plot_app?log=1b3c6d44-fd45-4044-a154-172aa87b2609

LorenzMeier, 2019-04-13:
Did you calibrate the airspeed sensor? If yes, then it should have been impossible to install the tubing incorrectly, which seems to be the root cause here. You have negative airspeed.

[image]

dannyfpv, 2019-04-16:
After some calibration attempts the air sensor was ready to be tested on flight, there no issues and worked correctly in mission mode.
[image]https://review.px4.io/plot_app?log=cbc1e07b-fbc9-43ac-9723-969d190265f0

closed 2019-04-24 by LorenzMeier