PX4/PX4-Autopilot #11761: Don`t work roslaunch px4 posix_sitl.launch

archee8, 2019-04-02:


bys1123, 2019-04-04:
[image]
I tried master branch yesterday, works well.

archee8, 2019-04-05:
I tried master branch but in firmware directory dont have iris.sdf. I copy iris.sdf from old version firmware but i get error than launch posix_sitl

lamping7, 2019-04-05:
How did you build the firmware? The model gets generated when you build the `sitl_gazebo` submodule. There are different make commands that do this depending on what you're trying to do. But, basically you need to build `sitl_gazebo`. `make px4_sitl_default sitl_gazebo` will build it without launching gazebo.

archee8, 2019-04-05:
i use this command from dev.px4.io
`make posix_sitl_default gazebo`
`source ~/catkin_ws/devel/setup.bash`    // (optional)
`source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default`
`export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)`
`export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo`
`roslaunch px4 posix_sitl.launch`

lamping7, 2019-04-05:
`posix_sitl_default` has been deprecated. Try `px4_sitl_default`. I can open an issue to fix the docs. Thanks for pointing it out.

archee8, 2019-04-06:
thank you! everything is working!

closed 2019-04-06 by archee8