PX4/PX4-Autopilot #11764: Mission speed

julianoes, 2019-04-02:
In recent jMAVSim SITL tests using the [Dronecode SDK](https://github.com/Dronecode/DronecodeSDK/blob/develop/integration_tests/mission_change_speed.cpp) I found that the MAVLink command [MAV_CMD_DO_CHANGE_SPEED](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_CHANGE_SPEED) does not seem to have any effect.

No matter what the speed is set, the vehicle speed is always 5 m/s presumably the cruise velocity.

Firmware version: 0b548fa6849e7a608b4c120fb66a760e51806056

julianoes, 2019-04-02:
@bresch any idea why this happens?

dagar, 2019-04-02:
Do you recall when it last worked?

julianoes, 2019-04-02:
My guess it's limited to cruise velocity somewhere in position control or flight task but I couldn't find it yet.

bresch, 2019-04-03:
Right, the velocity is limited by the cruise speed parameter. I have to check how "DO_CHANGE_SPEED" is implemented in the mavlink receiver.

julianoes, 2019-04-03:
It's implemented in the navigator actually.

julianoes, 2019-04-03:
Not fixed properly yet.

julianoes, 2019-04-04:
@bresch did you find anything?

bresch, 2019-04-04:
I need to discuss with @MaEtUgR for a proper fix, hopefully tomorrow we can have something.

MaEtUgR, 2019-04-08:
Does this only happen with smooth mission enabled or always? Because #11772 only changed the smooth vel task which
1. is not the upstream default
2. gives us a hint where the problem is

julianoes, 2019-04-08:
This happens with default jMAVSim SITL, so whatever comes with that. I don't know what sort of smoothing goes on.

MaEtUgR, 2019-04-08:
There seem to be two problems:
- Mission speed does not get applied in the smooth mission (`MPC_AUTO_MODE 1`, default)
- Speeds above the cruise speed do not work in general also with `MPC_AUTO_MODE 0`.

MaEtUgR, 2019-04-09:
I did a test with DronecodeSDK and `make && build/default/integration_tests/integration_tests_runner --gtest_filter="SitlTest.MissionChangeSpeed"` and at this point I still get the correct values: https://github.com/PX4/Firmware/blob/d9f8e37288d8e34baac79b32d71862045946534b/src/lib/FlightTasks/tasks/Auto/FlightTaskAuto.cpp#L129

MaEtUgR, 2019-04-09:
@julianoes The DronecodeSDK test seems super picky:
```
[11:40:47|Info ] Mission status update: 2 / 4 (mission_change_speed.cpp:165)
[11:40:53|Warn ] speed check, should be: 3 m/s, actually: 3.06059 m/s (mission_change_speed.cpp:88)
[11:41:15|Debug] current: 3, total: 4 (mission_impl.cpp:1200)
[11:41:15|Info ] Mission status update: 3 / 4 (mission_change_speed.cpp:165)
[11:41:21|Warn ] speed check, should be: 8 m/s, actually: 8.07201 m/s (mission_change_speed.cpp:88)
```
but at least it succeeds if it's within +-1m/s.

julianoes, 2019-04-09:
The warning is misleading, that should be a pass overall.

PR #11772 (2019-04-03): FlightTasks: fix mission DO_CHANGE_SPEED
This fixes the issue where the  DO_CHANGE_SPEED command was ignored and
the drone always travelled at the MPC_XY_CRUISE velocity.

Tested in SITL, fixes #11764.
PR #11836 (2019-04-12): Fix mission speed
**Test data / coverage**
I tested using the DronecodeSDK with command `make && build/default/integration_tests/integration_tests_runner --gtest_filter="SitlTest.MissionChangeSpeed"` and the following configuration combinations:
- `MPC_AUTO_MODE` 0 and 1 for default line tracking and jerk-limited trajectory
- `MPC_XY_VEL_MAX` to 12m/s default (which is above the tested 8m/s) and on 6m/s

The speed gets correctly applied and correctly limited.

**Describe problem solved by the proposed pull request**
fixes #11764 

**Additional context**
#11772, #11811
closed 2019-04-12 by julianoes