PX4/PX4-Autopilot #11767: nxp_fmuk66-v3 rev C issues

dagar, 2019-04-02:
**Tested on NXP v3.x**

**Modes Tested** 
_Position Mode:_ Armed and took-off as expected, once in the air the vehicle was oscillating. 

_Mission Plan Mode (Automated):_ Once switched from position to mission vehicle gain altitude but did not start the mission then switched back to position but UAV did not respond and crashed. 

_RTL (Return To Land):_ Untested. 

**Notes** 
Vehicle took-off in position mode but was oscillating while in the air. 
The vehicle did not respond to RC commands after switching from mission mode back to position mode.

https://review.px4.io/plot_app?log=d9e76bc4-5c82-42c5-84bf-19c127e78016
@igalloway @davids5 @dagar

dagar, 2019-04-02:
[image]

dk7xe, 2019-04-03:
My experience is, especially when using 4S with the hovergames LJI500/fmuk66 to lower the PID's by 30%. Also on 3S we had one setup that had oszillations. If you use the old S500 frame still oszillations are mich more the issue.

davids5, 2019-04-03:
@dagar What is implicated by a "check bit" on the graph? Mag timeouts to the estimator?

igalloway, 2019-04-04:
What magnitude were the oscillations?
also wanted to note here we did have two prototype boards with badly soldered BMM150 parts (Bosch Magnetometer). Noting in case we think this could have been a factor here.

igalloway, 2019-04-04:
Also note that we have had a number of successful flights, including autonomous modes here, in Eindhoven, and in Hamburg.
@Tony3dr I did just receive new "RevC" frames, I will try to get one sent out to you monday.

davids5, 2019-04-04:
> What magnitude were the oscillations?
> also wanted to note here we did have two prototype boards with badly soldered BMM150 parts (Bosch Magnetometer). Noting in case we think this could have been a factor here.

Ok that is a good data point and could explain the issue. 

@Tony3dr Please do only very cautious and conservative testing until we can build trust on the new unit.

Tony3dr, 2019-04-04:
> Also note that we have had a number of successful flights, including autonomous modes here, in Eindhoven, and in Hamburg.
> @Tony3dr I did just receive new "RevC" frames, I will try to get one sent out to you monday.

Awesome thanks!

Tony3dr, 2019-04-04:
> Ok that is a good data point and could explain the issue.
>
> @Tony3dr Please do only very cautious and conservative testing until we can build trust on the new unit.

@davids5 sounds good David, we will continue Testing.

Junkim3DR, 2019-04-05:
**Description**
- Drone oscillated during position mode, once switched to altitude mode, drone behaved as expected (It did not change altitude and drifted slightly according to wind direction while standing still in the air), then switched back to position mode, vehicle started to drift, while drifting RC commands were not responding so vehicle crashed.

- Drone oscillated during stabilized mode then flipped and crashed.

**Log Files and Screenshots**
_Note:_ You will notice at the very end of the flight, in Log: 3DR-MCa3ebe56-m01-NXP-02-01, there is a change from Yellow (Altitude) to Blue (Stabilized) but that switch was not made by the pilot; similar issue can be seen in Log: 3DR-MCa3ebe56-m01-NXP-02-02, a switch from Blue (Stabilized) to Purple (Mission) but this switched were not triggered by the pilot.

Position and Altitude modes test: https://review.px4.io/plot_app?log=4913ff83-05ca-4e01-89b8-e44fcfda8ded

Stabilized mode test: https://review.px4.io/plot_app?log=fc07658f-f940-4547-bbab-d55addbce5c1

**Vehicle**
- Frame: S500
- Flight Controller: NXP_FMUK66_V3

davids5, 2019-04-06:
@dagar can we review the flight logs on this next week?

LorenzMeier, 2019-04-13:
@dk7xe @igalloway Could you please re-test and comment what the current status of the board is? Could you please provide flight logs for a full mission?

davids5, 2019-04-17:
Also see [QGC not able to flash NXP FMUK66 #7360](https://github.com/mavlink/qgroundcontrol/issues/7360)

closed 2019-04-24 by LorenzMeier