PX4/PX4-Autopilot #11770: Periodic hardfault on Pixhawk 2 cube with thermal cal logging bit set

jzazbert, 2019-04-02:
**Describe the bug**
Modified PX4 1.8.2 with the thermal cal logging bit set, we've seen two different aircraft hardfault after ~15 hours, and then hardfaulted every ~8.5 hours after until finally hanging. All hardfault logs list sem_post.c line 118 as the source of error, with several different modules being listed as the cause. 

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on
2. Set sd logging to from boot until shutdown
3. Set sd log profile to 7 (default, EKF, thermal cal)
4. Reboot
5. Leave running for 24 hours
6. Note fault logs generated on SD card

**Expected behavior**
We expect it to not hardfault, and are concerned that the faults could show up worse in some situations. 

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

**Drone (please complete the following information):**
- Octo X running modified PX4 1.8.2 firmware on a Pixhawk 2.1 cube, with custom adapter board. HERE2 GPS, custom ESC and motors.

Attached files are the hardfault logs of the aircraft. The first fault happened about 15 hours in, and after that if faulted 5 more times roughly every 8.5 hours. In each case it rebooted, except the last one (the empty log) and the aircraft appeared to be frozen. 

[2019-04-01_83thermalchamber.zip](https://github.com/PX4/Firmware/files/3036226/2019-04-01_83thermalchamber.zip)

**Additional context**
This has happened with same firmware on two different cubes. It does not seem to happen if the aircraft is armed and running.

julianoes, 2019-04-03:
@jzazbert thanks for the report. Could you please attach the elf file that you flashed and a logfile matching it? That way we can use gdb to try to find out more.

julianoes, 2019-04-03:
Also, can you reproduce the issue with an unmodified firmware?

LorenzMeier, 2019-04-03:
We would need the source code to debug this or to reproduce this with unmodified firmware. We're not investigating anything that doesn't come with source as a matter of principle.

jzazbert, 2019-04-03:
We are going to test now with unmodified firmware on multiple hardware platforms.

julianoes, 2019-04-03:
This actually looks like #8913.

jzazbert, 2019-04-03:
As an update, we ran the hardware without the thermal cal logging and still had the fault behave the same. Testing now with 1.8.2 stock firmware.

closed 2019-07-16 by stale[bot]