PX4/PX4-Autopilot #11778: VTOL Pusher starts if autopilot boots with kill switch engaged

RomanBapst, 2019-04-03:
Discovered on the FunCub VTOL airframe.

RomanBapst, 2019-04-03:
Master commit ID: 6c1399b328ba2f2947

dagar, 2019-04-09:
@RomanBapst were you able to try this again with newer master (after April 4th)?

RomanBapst, 2019-04-10:
@dagar Tried on latest master, still the same issue.

How to reproduce 100%:
1) Power the drone
2) Engage kill switch
3) Power cycle the drone while kill switch is engaged (reboot via QGC does not work to reproduce)
4) Once autopilot finished booting disengage kill switch and notice pusher going to 1500PWM

Tests were done on Pixhawk4.
Last time tested on this commit: b6120f39f3cc27ce2ec1f3159d393e70db3f5ca5

RomanBapst, 2019-04-11:
@dagar @davids5 I actually think that this might not be caused by a bug but rather by this particular ESC used on the funcub. I compared the pwm outputs of the pusher channel in both situations (power cycling with kill switch on and power cycling with kill switch off) and the result it the same. When the kill switch is engaged the PWM is 950 and when disengaged it is 1000. I believe that when you power cycle with kill switch engaged the ESC receives a 950 PWM signals and thus it "calibrates" to this value.

Here is a log from boot with the kill switch engaged:
https://logs.px4.io/plot_app?log=8185cee6-3a7d-43ec-8a2a-ed5da3d5dcfa

I also tested this on another VTOL with a different ESC and everything worked fine, regardless of the kill switch position.

PR #11780 (2019-04-03): drivers/px4fmu initialize all arrays
 - fixes #11778

@RomanBapst could you give this a try on the FunCub (carefully)?
closed 2019-04-13 by LorenzMeier