PX4/PX4-Autopilot #11785: Master 1.9 notes (Prior to release)

ryanjAA, 2019-04-04:
We’ve been flying master for the last month or two and logged a lot of hours on this so I thought I’d write up some notes prior to release.

Just for clarity:

- We used 3 different airframe types.
- Flew in no wind up to 12 m/s+
- Conducted countless automissions from landing to takeoff and tested a lot of the new additions.
- Only tested fixed wing.

A few thoughts:

* Early landing (FW_LND_EARLYCFG)

	- We tested this a lot. It is great but it should not be enabled as default. A few reasons, if you have a plane that can create significant drag with insufficient thrust, you run a definite increased risk of crashing.
 	 - We saw this happen on all three different airframe types during our tests. No crashes to be clear occurred, but in some situations it put the planes past any comfort zone and borderline needed intervention. Of importance, each plane used wasn’t truly ideal as they all had oversized motors. 
	- Secondly, wind. If there is excessive wind and you are loitering in a higher drag setup, this it is not good. We actually clipped a tree top with a foam plane while being blown down a bit since there was so much lest authority, luckily it was just a wing nick and it kept going and landed fine (autonomously) but this feature coupled with adequate wind plus a not too powerful setup (or even normal power setup) can put you in a difficult spot and if you are a newer pilot this could cause problems, ie crash. 
	- Lastly is that while you tend to have increased airspeed while descending from the previous waypoint, the issues becomes that while actually conducting the loiter, that speed is bled off quickly and you end up back at point 1, potentially in a scenario where you have increased drag, maybe insufficient thrust and less authority. 

I want to make sure I don’t want to take away from the hard work that went into this, when it works it is great and really does assist with the flap deployment altitude bump but over the last 1+ months of testing, the one clear thing is that there were multiple times where we would definitely not want this enabled (and did disable it) and having this as standard on the next release will give people sense that it should be ok in any scenario. It is not. I strongly recommend disabling this as standard and having something in the docs about when to use it, its concerns, things to think about in wind, you specific power setup, etc. Prudence would dictate disabling this as standard.

* Runway tracking because of ground speed scaling of outputs. The ground speed scaling diminishes nose wheel control which is only evident if there are very long takeoffs (rollout). Generally, we ran three planes off the runway which is something we’ve never seen. This should be noted, explored, etc. I also think the reason why the takeoffs were so long should be assessed (brings me to the next point).

* Negative pitch setpoint - If you have a negative pitch setpoint to use during level flight, you are essentially wanting the plane to nose down (or up) to not change altitude while in the air. The issue is that during takeoff if you look at pitch setpoint you will see that it is actually the specified pitch minus the setpoint (we have takeoff pitch set to 12º but are getting 7º of pitch up on takeoff because we have a negative set point of 5 degrees (FW_PSP_OFF). What we are seeing is a real “airframe” pitch of 7 degrees instead of 12 (because of the offset and it thinks it is 12º) and the elevator up is sort of giving 12º up but not really since when level on the ground the elevator is technically giving 5º pitch down.
    - Quick solution is probably to simply let runway takeoff pitch setpoint (RWTO_PSP) able to be a positive number. It is currently not allowed to be.

* Wind - this is an older issue but still present. In higher winds - 10-15 m/s+ the TAS test ratio is constantly >1. @Dagar suggests increasing the TAS Gate size (EKF2_TAS_GATE) but I think there should be some talk of this prior. 

Landing itself needs a lot of work to get inline with a true and robust plane style landing.

Also, and this is just an observation, the ability to have control over the throttle time constant is good (FW_LND_THRTC_SC). 

Just wanted to share some thoughts.

LorenzMeier, 2019-04-12:
@dagar We've taken care of all but these.

dagar, 2019-04-13:
PR to disable FW_LND_EARLYCFG by default. https://github.com/PX4/Firmware/pull/11843

dagar, 2019-04-13:
> Runway tracking because of ground speed scaling of outputs. The ground speed scaling diminishes nose wheel control which is only evident if there are very long takeoffs (rollout). Generally, we ran three planes off the runway which is something we’ve never seen.

Is this a regression? Could you share any new logs that show the problem? Even more helpful would be if you have older logs where it wasn't an issue.

dagar, 2019-04-13:
PR to handle the FW pitch setpoint offset properly in all scenarios. https://github.com/PX4/Firmware/pull/11845

dagar, 2019-04-13:
> Runway tracking because of ground speed scaling of outputs. The ground speed scaling diminishes nose wheel control which is only evident if there are very long takeoffs (rollout). Generally, we ran three planes off the runway which is something we’ve never seen.

https://github.com/PX4/Firmware/blob/d36b06f779d49d8dc7387bf2b4fa3aade999ada3/src/modules/fw_att_control/FixedwingAttitudeControl.cpp#L647-L650

The current groundspeed scaling logic only applies once your groundspeed exceeds 60% of your min airspeed. I don't know the history of this magic number.

LorenzMeier, 2019-04-13:
@priseborough Could you please chime in here?

ryanjAA, 2019-04-14:
Here is some data and notes. They are grouped into tests and test days. Some brief notes are present but theres some flights to look at if need be for data. More if needed.

Test Plane   2

Flying fine but pre-loiter fix - take off is fast (no ground steering issues). No psp-offset:
https://review.px4.io/plot_app?log=6338f82c-163a-4250-bd03-9a3618d2fd22
https://review.px4.io/plot_app?log=d430c100-0dee-486b-857c-32670cbccfd8
https://review.px4.io/plot_app?log=05f3060a-7222-4423-bf19-05f02fef815b
https://review.px4.io/plot_app?log=e421d610-1f31-4f15-9321-8656a4d1ce8d
https://review.px4.io/plot_app?log=cbae2d66-5c98-4fd8-b6a6-b556d0591f0a
https://review.px4.io/plot_app?log=b3f64035-ab96-47c5-8c55-29df9ee412b9

Flying fine post-loiter fix - take off is fast (no ground steering issues). No psp-offset
https://review.px4.io/plot_app?log=72823f80-5c7c-4410-92c0-60c08fc66e70
https://review.px4.io/plot_app?log=5d6a9598-d08b-43a1-aa2d-3376618247e0
https://review.px4.io/plot_app?log=7407282f-7a21-4c55-aa5b-6b97c1d00f78
https://review.px4.io/plot_app?log=ea1cd724-576a-4bab-8ae4-dc37fc5a7e75

Flying fine - ~7 m/s wind creates TAS checkbit fail - take off is fast (no ground steering issues). No psp-offset
https://review.px4.io/plot_app?log=32c5651f-76a1-43bf-9e2b-86786b673142
https://review.px4.io/plot_app?log=500d30df-83ce-4298-8859-1dc0c80cc912
https://review.px4.io/plot_app?log=325f67ed-b7b5-40ac-9468-5cdeeae687da
https://review.px4.io/plot_app?log=0b2a1298-bb69-4895-98e8-517a823f26c0
https://review.px4.io/plot_app?log=82bd73cf-09f4-4a96-9954-13c849de8712

Flying fine -Testing FW_LND_THRTC_SC and altering landing settings (flare/Take-off is fast (no ground steering issues). No psp-offset
Lots of TAS flags for no real reason.

https://review.px4.io/plot_app?log=e98c02d4-2bdd-4ee4-9d83-000f71c6f829
https://review.px4.io/plot_app?log=c150c0b6-c1b7-4b7a-8ef5-9cb7c221424d
https://review.px4.io/plot_app?log=ca05c861-63da-4999-b20c-d7c486510757
https://review.px4.io/plot_app?log=a70dfbed-defb-4151-bee7-78833d2b7d32
https://review.px4.io/plot_app?log=b57515e2-db1b-4aa2-885d-54143f09b3e4
https://review.px4.io/plot_app?log=2398c72f-da49-49bf-bdfb-28f9fba7c297
https://review.px4.io/plot_app?log=e7b0753d-e7db-4312-a676-68b1c394cd9a
https://review.px4.io/plot_app?log=349097d9-e4d0-4a1b-b1a1-fe3ceb6e300b
https://review.px4.io/plot_app?log=1a017f5a-3958-4ff8-a2cf-d289a8963c60
https://review.px4.io/plot_app?log=99152b04-b769-4f64-aa23-db466b113c48

Test plane 3

CRASH - LOG missing (will need to locate). Due to ground steering, shallow takeoff and then pilot error   switching modes.

12-14 m/s wind.   Flying fine but TAS checkbit error.
https://review.px4.io/plot_app?log=1c6d0251-272d-4434-b898-d109b0b13ab2 Ground Steering failure (high head wind and 60% issue most likely)
https://review.px4.io/plot_app?log=b2f50bec-a9d3-41ae-a1b1-3bdd701ec2c9 Ground Steering failure (high head wind and 60% issue most likely)
https://review.px4.io/plot_app?log=222b48b8-54bc-49af-97af-39eaadc93bae Ground Steering failure (high head wind and 60% issue most likely) Increased FW_PR_P leads to oscillations on FFT graph - flying sufficiently though.

Testing flare to alt WITHOUT early landing on.
https://review.px4.io/plot_app?log=51293785-52c3-4901-a88f-327aa675459c Less wind (5-7 m/s) lots of checkbits hitting.
https://review.px4.io/plot_app?log=21ed6aaa-4bcd-4542-befe-0f407e7c35f8 Lands very long (too long)

Lower wind, testing decreased FW_PR_P
https://review.px4.io/plot_app?log=14e3a6cf-7688-4c77-a5bc-44c4099c1782 High vibration - motor change needed but not done for consistency

Poor take off and landing
https://review.px4.io/plot_app?log=bd117170-1dca-42a8-a309-58b537296419 Steering error and shallow takeoff so position mode entered into and then back to mission. Mag checkbits may be the issue here.
https://review.px4.io/plot_app?log=3198efbc-5eff-4d7a-8371-31aa64c563b6 Long landing, put into position mode to cut motor right when about to land.
https://review.px4.io/plot_app?log=08ff0f58-b264-49b0-a694-58eab9b76ffd Super long landing, ended  landing while just about to land as was going to hit tree line, avoided crash by turning right.
https://review.px4.io/plot_app?log=998a6802-6a13-41d2-95ba-15d5213d0e03 Ground steering so bad that turns left to circle back to hit takeoff waypoint. Mag checkbits?? Wind 7-9 m/s

Very good flights and landings. 12 m/s wind. Takeoff and ground steering good.
https://review.px4.io/plot_app?log=652d869f-713b-424a-8402-8d3a233a42ab Really good flight and landing - mode change was due to new ground-station improperly calibrated for mode change switch.
https://review.px4.io/plot_app?log=b16d8104-d0e3-4e84-8ed4-656989e639fa Really good flight and landing - mode change was due to new ground station improperly calibrated for mode change switch.

Lastly is a snapshot from one of our daily flight assessments post flight. Done to see impact on flare at touchdown. Any questions and let me know.

[image]

ryanjAA, 2019-04-14:
I think the 60% scaling issue is an issue for two reasons, one if there is high head wind on takeoff, where on one hand would lead to shorter takeoffs and not pose an apparent issue but the issue is still there just not realized all the time and secondly (and more importantly), when you are at 60% of min speed and not at whatever percentage you have set for pitch up to start (we are at 85% of min speed) that 25% can be super fast but if it takes some time, (seconds, meters, however you want to measure it), it poses an issue or window for there to be an error present. 
Obviously the concept makes sense, as speed is increased we dont want a hard right on the nose wheel so we dont flip over but there is probably something that needs to be done.

Also, we saw on many occasions, when hitting the mission start the nose wheel would immediately go full left or right for a split second. On the smaller planes this would cause a flip over and the mission would need to be aborted (likely not in the above logs as those are all flights), but in some instances it would still recover quickly and be okay (no flipping, etc). On the larger planes it occurred but would recover very fast all but a few times. This probably is reflected in the above. Why, i'm not sure but we ended up limiting the max and min pwm the nose servo could output via qgc, not the servo programmer. Food for thought. There will be examples of both in the above.

priseborough, 2019-04-17:
> * Wind - this is an older issue but still present. In higher winds - 10-15 m/s+ the TAS test ratio is constantly >1. @dagar suggests increasing the TAS Gate size (EKF2_TAS_GATE) but I think there should be some talk of this prior.
> 

I've been through the https://review.px4.io/plot_app?log=1c6d0251-272d-4434-b898-d109b0b13ab2 log with the following observations:

1) Sideslip innovations are small, but airspeed innovations are large which normally indicates an issue with airspeed sensor calibration:

[image]
 
2)  This is confirmed by over-plotting GPS ground speed and true airspeed which for a close circuit flown should show TAS sitting halfway between min and max ground speed:

[image]

The continued high TAS innovation test ratio looks to be a true indicator of a large pitot-static calibration error.

ryanjAA, 2019-04-17:
Thanks for looking and good to know. It does raise questions though as every one of these flights are done on the sensirion sdp33 units all with measured tubes, etc and calibrated prior to each flight (or groupings of flights).

That raises questions as to what to do then.

priseborough, 2019-04-17:
The other high wind logs show similar characteristics, so this flight vehicle definitely needs to have the differential pressure scale factor adjusted.

[image]

[image]

A calibrated probe/pressure sensor by itself is insufficient - the airflow around the aircraft induces a change in static pressure at the static holes and therefore the differential pressure seen by the sensor. This varies from installation to installation and measuring and correcting for it should be part of the commissioning procedure. 

I have never commissioned PX4 for a FW platform so am not familiar with the parameters used to calibrate the airspeed sensor, however looking in the code for the SPD3X driver I do not see where the user can calibrate for this error. I will investigate further.

ryanjAA, 2019-04-17:
Thanks for the insight on this. The pitots are in the nose of the aircraft protruding out their full length, ie not on the wings or sides of the fuselage. This has always been in an attempt to have a clean access to airflow.

This does raise an interesting point of how to measure and correct for this in any installation. I do have some high wind logs of this occurring with a different airspeed sensor and if keeping the pitot out the nose away from everything isn’t ideal than figuring out what is or how we correct for that would probably be pretty helpful for all. 

Thanks for looking into this further. I’m quite intrigued.

tstastny, 2019-04-17:
> A calibrated probe/pressure sensor by itself is insufficient - the airflow around the aircraft induces a change in static pressure at the static holes and therefore the differential pressure seen by [...]

@ryanjAA @priseborough Just saw this in passing and thought I might give some heads-up from my own experience with this sensor -- we use it exclusively because it's great at low speeds, but Paul is correct that the given parameters do not account for other flow around the aircraft. We've dealt with this so far by simply using one of the parameters (specifically the diameter) to adjust (fudge) this after checking the performance in flight. This is quite simple to do by just flying a constant altitude loiter for a few times in a row, and then adjusting the param until the true airspeed sits directly in the center of the ground speed oscillations -- I've actually written a script that will automatically do this using the wind triangle equations and a constant wind assumption to back calculate out the diameter that fits the physics using nonlinear least squares (only a few lines of matlab code -- but I'm sure could also easily be done in python). It's a little hacky, as of course a more proper solution would be to introduce a new parameter for airflow interference effects .. but until then, adjusting the diameter is pretty easy/effective.

Anyway - just thought it may be helpful!

priseborough, 2019-04-17:
@ryanjAA Air in front of the nose has been partially slowed which increases static pressure (Bernoullis law applies) and reduces differential pressure/airspeed measured. Air above the wing or around the side/topof the fuselage tends to be accelerated and therefore has lower than atmospheric pressure which causes the airspeed to read high. Probes on full size aircraft are carefully placed in a region where the air is close to free-stream velocity and static pressure close to atmospheric.

Flying in a circular loiter and adjusting the scale factor until the average airspeed innovation is zero or the TAS is in the middle of the ground speed is one way to do it. Also be aware of TAS vs EAS/IAS differences as test altitude and temperature increases.

ryanjAA, 2019-04-17:
Interesting, although that does raise some other questions given i've seen this on several airframe types, not the same vehicle or even vehicle shape which means there needs to be a way to manage this other than putting something at risk in the air. Granted, i've never seen a crash because of this but if the front of the fuse is going to give slower than actual readings and placement around the wing area gives faster than actual then the safe bet is still the nose. I can pull out all the CFD work and modeling done etc but that's not helpful to other users of other airframe types which is the real aim here to solve. Surely I can't be the only person seeing this. Maybe only very few are reporting it or know to even report it? @tstastny definitely has seen it given your comments so I'm not alone here but the sdp3x units are only on our test vehicles and I have some logs of this happening on the older ms4525 airspeed as well. A fix with those would be I assume altering the SENS_DPRES_OFF for non sdp33 units? 

We ran some bench tests (indoors) with a set known windspeed device and an anemometer as well as the sdp33 units in the plane prior to the first real tests as a sanity check but I believe we only ran them to 10-12 MPH so much slower and not in a real-world scenario (as there was no reason to doubt their use). 

@tstastny is your work private or available somewhere? I'm wondering if this can be incorporated in during flight (well instead of fudging the tube mm, adjust the scale factor) but there must be a methodical way to approach this rather than a fly and test and fly and test given it's one of the critical things to know and from the sounds of, people are possible having this issue and not even knowing it. 

The fact that TAS vs IAS will change at altitude and temp (is a known understandable thing) but the fact that it further convolutes the issue means this all probably really needs some attention.

Antiheavy, 2019-04-17:
sdp3x sensors are "flow-through" style sensors and can be tricky to calibrate.  Amazing performance for their size, but very sensitive to pressure drop in the tubing and pitot-static tube upstream of the sensor.  Also can't test for leaks very easily since a flow-through system can't hold pressure.  We continue to use the old cheap ms4525 series even though they have relatively terrible performance.

closed 2019-08-07 by stale[bot]