PX4/PX4-Autopilot #11786: Pixhawk4 will not bootup with current master

bys1123, 2019-04-04:
**Describe the bug**
Pixhawk4  will not bootup with current master, all fmu led is off after bootloader.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash master firmware

**Additional context**
I tried flash Ardupilot for reset parameters, an re-flash, it's not work either.

bkueng, 2019-04-04:
Same for me. #11769 seems to be responsible, before that it boots.
@davids5 can you have a look?

Console output:
```
up_assert: Assertion failed at file:irq/irq_unexpectedisr.c line: 65 task:
```

davids5, 2019-04-04:
@bkueng @bys1123 - Thank you for reporting the issues.  I have fixed it here https://github.com/PX4/Firmware/pull/11791

closed 2019-04-04 by dagar