PX4/PX4-Autopilot #11787: Permanent slope during transition

DavidBouSaba, 2019-04-04:
Hello,

We use a PX4 quad X tailsitter configuration and we have the following issue:

For all the transitions tests we made (manual or missions you can see a picture below), the aircraft was never finishing the transition (staying dark pink and never going to light pink on the log meaning full fw mode not reach) and having a stable maintained pitch angle at the end of this transition of 90-74,5 = 14,5°).

[image]

We please have 2 questions:

- Why the transition is not fully accomplished?
- Where this angle of 14,5° could come from (we have no offset or parameter with this value (the picth offet we have is set to 0)?

Thanks in advance,

julianoes, 2019-07-04:
Sorry for the late response. Is this still an issue? And if yes, could you add link to a log?

julianoes, 2019-10-09:
Closing because there is no log so far.

closed 2019-10-09 by julianoes