PX4/PX4-Autopilot #11789: Ubuntu 16.04 px4 Compiler Environment problem

jumppp, 2019-04-04:
I just got in touch with pixhawk and I am configuring it in the development environment under Ubuntu.I tried it many times, including using the official method, but it still failed.I have been troubled by this problem for a long time, how should I solve it?

jump@jump:~/src/Firmware$ make px4fmu-v2_default
Makefile:212: px4fmu-v2_default has been deprecated and will be removed, please use px4_fmu-v2_default!
make px4_fmu-v2_default
make[1]: Entering directory '/home/jump/src/Firmware'
[ 0%] Built target git_nuttx_apps
[ 0%] Generating mixer_multirotor_6dof.generated.h
[ 0%] Built target git_nuttx
[ 0%] Built target uorb_headers
[ 0%] Generating mixer_multirotor.generated.h
[ 0%] Built target ver_gen
[ 0%] Built target git_gps_devices
Scanning dependencies of target git_ecl
[ 0%] git submodule src/lib/ecl
Scanning dependencies of target git_mavlink_v2
[ 1%] git submodule mavlink/include/mavlink/v2.0
[ 1%] Built target git_ecl
[ 1%] Built target git_mavlink_v2
[ 1%] Generating mixer_multirotor_normalized.generated.h
[ 1%] Built target mixer_gen_6dof
[ 1%] Built target mixer_gen
Scanning dependencies of target nuttx_c_build
Scanning dependencies of target nuttx_sched_build
Scanning dependencies of target nuttx_fs_build
Scanning dependencies of target nuttx_drivers_build
[ 1%] Copying NuttX/apps to build/px4_fmu-v2_default/NuttX
[ 1%] Copying NuttX/apps to build/px4_fmu-v2_default/NuttX
[ 1%] Copying NuttX/apps to build/px4_fmu-v2_default/NuttX
[ 2%] Copying NuttX/apps to build/px4_fmu-v2_default/NuttX
[ 2%] Copying NuttX/nuttx to build/px4_fmu-v2_default/NuttX
[ 2%] Copying NuttX/nuttx to build/px4_fmu-v2_default/NuttX
[ 2%] Copying NuttX/nuttx to build/px4_fmu-v2_default/NuttX
[ 2%] Copying NuttX/nuttx to build/px4_fmu-v2_default/NuttX
[ 2%] Generating nuttx/configs/dummy/Kconfig, nuttx_config_kconfig.stamp
[ 2%] Generating nuttx/configs/dummy/Kconfig, nuttx_config_kconfig.stamp
[ 2%] Generating nuttx/configs/dummy/Kconfig, nuttx_config_kconfig.stamp
[ 2%] Generating nuttx/configs/dummy/Kconfig, nuttx_config_kconfig.stamp
[ 2%] Copying NuttX config nsh
[ 2%] Copying NuttX config nsh
[ 2%] Copying NuttX config nsh
[ 2%] Copying NuttX config nsh
[ 2%] Copying NuttX compressed config nsh and inflating
[ 2%] Copying NuttX compressed config nsh and inflating
make[5]: warning: jobserver unavailable: using -j1. Add '+' to parent make rule.
[ 3%] Copying NuttX compressed config nsh and inflating
[ 3%] Copying NuttX compressed config nsh and inflating
make[5]: warning: jobserver unavailable: using -j1. Add '+' to parent make rule.
make[5]: warning: jobserver unavailable: using -j1. Add '+' to parent make rule.
make[5]: warning: jobserver unavailable: using -j1. Add '+' to parent make rule.
[ 3%] Generating nuttx/include/nuttx/version.h, nuttx/include/nuttx/config.h
[ 3%] Generating nuttx/include/nuttx/version.h, nuttx/include/nuttx/config.h
make[5]: warning: jobserver unavailable: using -j1. Add '+' to parent make rule.
[ 3%] Generating nuttx/include/nuttx/version.h, nuttx/include/nuttx/config.h
make[5]: warning: jobserver unavailable: using -j1. Add '+' to parent make rule.
make[5]: warning: jobserver unavailable: using -j1. Add '+' to parent make rule.
[ 4%] Generating nuttx/include/nuttx/version.h, nuttx/include/nuttx/config.h
make[5]: warning: jobserver unavailable: using -j1. Add '+' to parent make rule.
/bin/sh: 1: tools/mkversion: Text file busy
make[5]: *** [include/nuttx/version.h] Error 2
NuttX/CMakeFiles/nuttx_sched_build.dir/build.make:363: recipe for target 'NuttX/nuttx/include/nuttx/version.h' failed
make[4]: *** [NuttX/nuttx/include/nuttx/version.h] Error 2
make[4]: *** Deleting file 'NuttX/nuttx/include/nuttx/version.h'
CMakeFiles/Makefile2:8409: recipe for target 'NuttX/CMakeFiles/nuttx_sched_build.dir/all' failed
make[3]: *** [NuttX/CMakeFiles/nuttx_sched_build.dir/all] Error 2
make[3]: *** 正在等待未完成的任务....
/bin/sh: 1: tools/mkversion: Text file busy
make[5]: *** [include/nuttx/version.h] Error 2
NuttX/CMakeFiles/nuttx_c_build.dir/build.make:734: recipe for target 'NuttX/nuttx/include/nuttx/version.h' failed
make[4]: *** [NuttX/nuttx/include/nuttx/version.h] Error 2
make[4]: *** Deleting file 'NuttX/nuttx/include/nuttx/version.h'
CMakeFiles/Makefile2:8376: recipe for target 'NuttX/CMakeFiles/nuttx_c_build.dir/all' failed
make[3]: *** [NuttX/CMakeFiles/nuttx_c_build.dir/all] Error 2
make[5]: *** [arch/arm/src/board] Error 1
NuttX/CMakeFiles/nuttx_drivers_build.dir/build.make:611: recipe for target 'NuttX/nuttx/include/nuttx/version.h' failed
make[4]: *** [NuttX/nuttx/include/nuttx/version.h] Error 2
make[4]: *** Deleting file 'NuttX/nuttx/include/nuttx/version.h'
CMakeFiles/Makefile2:8636: recipe for target 'NuttX/CMakeFiles/nuttx_drivers_build.dir/all' failed
make[3]: *** [NuttX/CMakeFiles/nuttx_drivers_build.dir/all] Error 2
make[5]: *** [arch/arm/src/board] Error 1
NuttX/CMakeFiles/nuttx_fs_build.dir/build.make:301: recipe for target 'NuttX/nuttx/include/nuttx/version.h' failed
make[4]: *** [NuttX/nuttx/include/nuttx/version.h] Error 2
CMakeFiles/Makefile2:8474: recipe for target 'NuttX/CMakeFiles/nuttx_fs_build.dir/all' failed
make[3]: *** [NuttX/CMakeFiles/nuttx_fs_build.dir/all] Error 2
Makefile:105: recipe for target 'all' failed
make[2]: *** [all] Error 2
Makefile:190: recipe for target 'px4_fmu-v2_default' failed
make[1]: *** [px4_fmu-v2_default] Error 2
make[1]: Leaving directory '/home/jump/src/Firmware'
Makefile:212: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2

julianoes, 2019-04-04:
@jumppp thanks for the report, I think this is the same as #11003.

Can you try to install `ninja-build`, and then try again after a `make clean`?

jumppp, 2019-04-05:
i installed ninja-build , The compilation result is as follows. Is this compiled successfully?

jump@jump:~/src/Firmware$ make px4fmu-v2_default
Makefile:212:  px4fmu-v2_default has been deprecated and will be removed, please use px4_fmu-v2_default!
make  px4_fmu-v2_default
make[1]: Entering directory '/home/jump/src/Firmware'
-- PX4 config file: /home/jump/src/Firmware/boards/px4/fmu-v2/default.cmake
-- PX4 config: px4_fmu-v2_default
-- PX4 platform: nuttx
-- PX4 lockstep: disabled
-- PX4 version: v1.9.0-beta2-2-g7a5aaa3
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.3.1
-- The C compiler identification is GNU 7.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-672-gf95cd4b
-- Building and including px4_io-v2_default
-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jump/src/Firmware/build/px4_fmu-v2_default
[1/673] git submodule platforms/nuttx/NuttX/nuttx
[7/673] git submodule platforms/nuttx/NuttX/apps
[10/673] git submodule src/lib/ecl
[15/673] git submodule src/drivers/gps/devices
[21/673] git submodule mavlink/include/mavlink/v2.0
[116/673] Performing configure step for 'px4io_firmware'
-- PX4 config file: /home/jump/src/Firmware/boards/px4/io-v2/default.cmake
-- PX4 config: px4_io-v2_default
-- PX4 platform: nuttx
-- PX4 lockstep: disabled
-- PX4 version: v1.9.0-beta2-2-g7a5aaa3
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.3.1
-- The C compiler identification is GNU 7.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-672-gf95cd4b
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jump/src/Firmware/build/px4_fmu-v2_default/external/Build/px4io_firmware
[154/673] Performing build step for 'px4io_firmware'
[1/199] git submodule platforms/nuttx/NuttX/nuttx
[8/199] git submodule platforms/nuttx/NuttX/apps
[199/199] Creating /home/jump/src/Firm...d/px4io_firmware/px4_io-v2_default.px4
[673/673] Creating /home/jump/src/Firm..._fmu-v2_default/px4_fmu-v2_default.px4
make[1]: Leaving directory '/home/jump/src/Firmware'
jump@jump:~/src/Firmware$

julianoes, 2019-07-05:
Presumably fixed in #11821.

closed 2019-07-05 by julianoes