PX4/PX4-Autopilot #11794: Continuous "Data link lost" messages in SITL

julianoes, 2019-04-05:
I'm ran into the case where I get continuous "Data link lost" messages in SITL, using the Dronecode SDK and QGC.

```
WARN  [commander] Data link lost
WARN  [commander] Data link lost
WARN  [commander] Data link lost
WARN  [commander] Data link lost
WARN  [commander] Data link lost
WARN  [commander] Data link lost
WARN  [commander] Data link lost
WARN  [commander] Data link lost
WARN  [commander] Data link lost
```

```
[10:39:25|Debug] MAVLink: critical: Data link lost (system_impl.cpp:292)
[10:39:25|Debug] MAVLink: info: Data link regained (system_impl.cpp:292)
[10:39:26|Debug] MAVLink: critical: Data link lost (system_impl.cpp:292)
[10:39:26|Debug] MAVLink: info: Data link regained (system_impl.cpp:292)
[10:39:27|Debug] MAVLink: critical: Data link lost (system_impl.cpp:292)
[10:39:27|Debug] MAVLink: info: Data link regained (system_impl.cpp:292)
[10:39:28|Debug] MAVLink: critical: Data link lost (system_impl.cpp:292)
[10:39:28|Debug] MAVLink: info: Data link regained (system_impl.cpp:292)
```

LorenzMeier, 2019-04-05:
I believe this has to do with the type of connection the SDK uses. I've heard this before. Look into how GCS timeouts are being used and what the SDK exactly is sending.

julianoes, 2019-04-05:
As far as I know the SDK is sending pretty much the same as QGC ust with another component ID.

The problem was that both mavlink instances keep sending the time when the last heartbeat arrived which caused the toggling behaviour in commander.

PR #11795 (2019-04-05): commander: fix toggling datalink lost and regained
When using QGC and/or the Dronecode SDK it was possible to get in a state where the two mavlink instances were both publishing their last heartbeat_time and cause commander to consistently toggle between data link lost and regained.

With this fix, we only ever look at the very last heartbeat time and therefore seem to avoid this issue.

Tested in SITL, fixes #11794.
closed 2019-04-05 by dagar