PX4/PX4-Autopilot #11798: PX4/ecl v1.9.0 release blockers

dagar, 2019-04-05:
- [ ] https://github.com/PX4/ecl/pull/578
 - [ ] https://github.com/PX4/ecl/pull/593
 - [ ] https://github.com/PX4/ecl/pull/505

dagar, 2019-04-05:
@priseborough @RomanBapst @mhkabir @CarlOlsson please update this issue with any outstanding PX4/ecl changes or issues needed before a v1.9.0 release.

priseborough, 2019-04-07:
https://github.com/PX4/ecl/pull/578 has been merged

https://github.com/PX4/ecl/pull/593 is awaiting test evidence from the author, but is otherwise a low risk change.

https://github.com/PX4/ecl/pull/505 is a more complex change and without proper testing including the revised logic for handling of optical flow timeouts could introduce bugs I am not comfortable with it being included in a stable release without significant testing.

dagar, 2019-04-08:
Thanks Paul.

@mhkabir @RomanBapst thoughts on testing https://github.com/PX4/ecl/pull/505 this week for release inclusion? Either way we should discuss an appropriate test plan.

LorenzMeier, 2019-04-13:
@RomanBapst Please help with testing where possible.

@dagar Any updates?

mhkabir, 2019-04-13:
Let's remove the optical flow PR from release. It will require more testing and work which I'm not able to contribute right now.

dagar, 2019-04-13:
We've arranged to get @PX4/testflights optical flow hardware (pmw3901), but I don't know the realistic time frame.

> Let's remove the optical flow PR from release. It will require more testing and work which I'm not able to contribute right now.

Sounds like we're bumping it for v1.9.0, but can we still try to get a test plan together for these items?

LorenzMeier, 2019-04-13:
@jkflying has done extensive testing of this - could you please pitch in here? We're flying this daily in Zurich.

dagar, 2019-04-13:
@mhkabir what about https://github.com/PX4/ecl/pull/593?

dagar, 2019-04-13:
> @jkflying has done extensive testing of this - could you please pitch in here? We're flying this daily in Zurich.

I've rebased https://github.com/PX4/ecl/pull/505 on current master. The branch ecl_PR-505 was created for testing PX4/Firmware side. https://github.com/PX4/Firmware/tree/ecl_PR-505

Binaries available here under artifacts. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-compile/detail/ecl_PR-505/1/pipeline

jkflying, 2019-04-14:
I'll give this a flight tomorrow if the weather is ok, I don't forsee any issues caused by https://github.com/PX4/ecl/pull/505 though since the pmw3901 runs at ~40hz.

jkflying, 2019-04-16:
@dagar should I test the exact binaries? I usually increase tx/rx buffer sizes a bit because of our particular setup, but I can disable our log streaming, avoidance etc if it is important to use the exact binaries.

dagar, 2019-04-16:
@jkflying no, it's just for convenience.

jkflying, 2019-04-16:
Flight with PMW3901 was good. No weird twitches, seemed smooth, did multiple takeoffs with optical flow straight from the grass, did a mission with avoidance enabled. Everything seemed fine. Merging is OK from my side.

mhkabir, 2019-04-16:
@jkflying Can you post the logs on the ECL PR please?

closed 2019-04-24 by LorenzMeier