PX4/PX4-Autopilot #11799: Bug: RC mode switch requires toggling

AuterionWrikeBot, 2019-04-05:
How to reproduce:
power off vehicle; put RC switch into position; power on vehicle --> vehicle will not go to position mode (unless you changed your defaults)

This is possibly a regression

@DanielePettenuzzo reported this

dagar, 2019-04-05:
@DanielePettenuzzo how are you expecting this to work?

bresch, 2019-04-07:
@dagar I think that's related to what we discussed on Friday: the drone should go into the mode defined by the mode switch at boot. Right now, it always initialized to "Manual" even if the switch is pointing to an other mode.

As discussed, I'll come with more details.

dagar, 2019-04-07:
Short term for this particular case we could add logic in commander to re-evaluate the mode switches if disarmed and the condition valid flags change.

To move beyond RC setups we need something different/better though.

dagar, 2019-04-07:
To expand on that a bit, the scenarios I'm interested in are 1) locked down configurations (likely without real RC) where modes like MANUAL, ACRO, STABILIZED aren't even accessible (not just hidden) and 2) configurations with no manual input at all.

For these I think we'll ultimately need to be able to set a default/startup mode, allowable modes, and think through the initialization (mostly estimator) from the user perspective rather than just leaving them in MANUAL.

MaEtUgR, 2019-04-09:
Yes I think short term we can just check the RC mode switch when the RC connects for the first time and maybe also when the RTL finished because taking off in RTL doesn't make sense. That would still cover the case you control the drone over the datalink/tablet which switches out of manual and and regaining RC during an RTL which is a reconnect and not the first RC connection.

@dagar What about having an initialization state which requires nothing, doesn't allow any flight and cannot be entered into again? That would prevent manual mode to show even though it's no option. Did you mean to also block the disallowed modes e.g. manual in failsafe situations?

thomasgubler, 2019-05-16:
@MaEtUgR is this fixed via https://github.com/PX4/Firmware/issues/11914 ?

MaEtUgR, 2019-05-23:
@thomasgubler Yes thanks, I wasn't aware this is still open.

closed 2019-05-23 by MaEtUgR