PX4/PX4-Autopilot #11800: PWM does not ramp during arming

nathanmcferrin, 2019-04-05:
Summary:
With v1.8.2 stable, pwm does not ramp during arming, but rather transitions abruptly from the disarmed value (900 in my case) to the minimum value (1100 in my case). The lack of ramp causes certain motors I'm using to stall on arming roughly 5-10% of the time. 

Steps used to produce the behavior:
--Using Cube (ProfiCNC v2.1) Pixhawk.
--Loaded v1.8.2 stable release via QGC. Note that I've also seen the same behavior when using a version built from the stable 1.8.2 built with make px4fmu-v3_default
--Using quad x configuration (sys_autostart = 4001)
Calibrated sensors, set circuit breakers as necessary to enable px4 to pass arm checks.
--Set logger to log from power-up to power-down. Logging default and high rate data, although you can see it in either.
--Armed via command line (commander arm)
--Waited several seconds
--Disarmed via command line (commander disarm)

Expected Behavior:
Based on a brief look through pwm_limit.c, I expected a 0.5 second ramp from disarmed to min value. This value appears to be coded as a hard definition in pwm_limit.h (RAMP_TIME_US 500000).

Log file available at: 
https://review.px4.io/plot_app?log=9f75b199-eb52-436a-ab92-b45447e77624

Screenshots of PWM commanded from this log:
[image]
[image]

Note that the log/screenshots were from an arm/disarm cycle done on my desk via console. This was not an actual flight.

LorenzMeier, 2019-04-05:
What is the result with current master / the development version?

nathanmcferrin, 2019-04-05:
Just tried with the current master / development version, with the same result.

dagar, 2019-04-06:
@nathanmcferrin could you share the output of `pwm info`?

nathanmcferrin, 2019-04-06:
With the current master / development firmware: 
[image]

dagar, 2019-04-06:
@nathanmcferrin where is the 2nd plot from?

[image]

nathanmcferrin, 2019-04-06:
Second plot is same data as the first plot, just zoomed in to more accurately show the rise time.

nathanmcferrin, 2019-04-08:
@LorenzMeier does the output of "pwm-info" indicate anything that may be disabling the PWM ramping?

julianoes, 2019-07-08:
FYI @MaEtUgR

parakhm95, 2021-01-14:
So, for me, I was able to avoid this ramp skipping by setting COM_PREARM_MODE variable to disabled. Then, the MIN to ARM value ramp works as expected when using IO board for PWM output. Have to verify for SBUS as well. Hope this helps someone out. 

This also relates to :  https://github.com/PX4/PX4-Autopilot/issues/15364 also