PX4/PX4-Autopilot #11812: ECL breaks gazebo when there is a lidar

AuterionWrikeBot, 2019-04-08:


thomasgubler, 2019-04-08:
@dagar @priseborough FYI

mrivi, 2019-04-08:
ecl commits that introduces the issue https://github.com/PX4/ecl/commit/938c8ad9ad63600240cffc7b394ad1abbe3d58e2

AuterionWrikeBot, 2019-04-09:
➤ Roman Bapst commented:

From what I have seen the lidar in the gazebo model was falsely reporting 3m on ground which was the reason why the estimator performed the mag alignement on ground (the patch mentioned by Martina Rivizzigno allows the terrain estimate to be used to determine if sufficient distance from the ground has been achieved to trigger a reset).
However, there is still no reason why this should trigger a divergence of the estimator.
I did not yet spend much time looking at the log but at a first glance it seemed like an issue with the covariance. The sensor data seems all fine.

AuterionWrikeBot, 2019-04-09:
➤ Roman Bapst commented:

Log with the patch (divergence):
https://review.px4.io/plot_app?log=f1ff8c5b-d47a-4065-8a6a-1beafacd1b6c

Log without the patch (no divergence):
https://review.px4.io/plot_app?log=faeb58bc-7b30-49da-ac33-7a1e7ba53645

AuterionWrikeBot, 2019-04-09:
➤ Roman Bapst commented:

Martina Rivizzigno Could you maybe provide another log without the patch (no divergence) but stay a bit longer on the ground landed? I think in the first log the drone is just on the ground while in the second one it actually flies.

mrivi, 2019-04-09:
@RomanBapst log with vehicle on the ground and no divergence https://review.px4.io/plot_app?log=1f35a099-1f52-40a0-aebc-86a9de9171bd

AuterionWrikeBot, 2019-04-09:
➤ Martina Rivizzigno commented:

ECL patch https://github.com/PX4/ecl/pull/596

AuterionWrikeBot, 2019-04-09:
➤ Martina Rivizzigno commented:

ECL patch merged, Firmware update submodule PR https://github.com/PX4/Firmware/pull/11817

dagar, 2019-04-09:
ECL update in https://github.com/PX4/Firmware/pull/11814

mrivi, 2019-04-10:
#11817 was merged. Closing this issue

closed 2019-04-10 by mrivi