PX4/PX4-Autopilot #11815: Incomplete log file from flight

kaklik, 2019-04-08:
We experienced several issues consisting of log file ending in the middle of the flight. 
As the example of this situation, I select the following two cases where we have the log from QGC, therefore it is proof that the PX4 log file is incomplete. 
I checked that log file at SDcard has the same size as downloaded one. Therefore it is not a copying issue. 

Example A: 
[PX4 log](https://logs.px4.io/plot_app?log=87bd808c-28bc-4e3e-8711-9dded2b2bb27)
[QGC log file 2019-03-31 16-54-20.tlog](https://github.com/PX4/Firmware/files/3056575/2019-03-31.16-54-20.log)

Example B
[PX4 log](https://logs.px4.io/plot_app?log=496b6cb5-5f70-43a4-99ea-4ea441213b7a)
[QGC log file 2019-04-07 16-41-41.tlog](https://github.com/PX4/Firmware/files/3056584/2019-04-07.16-41-41.log)

We also have an observation about log file has been generated in the folder from the previous day and has the wrong name according to time, although the GPS has a 3D fix and arming without GPS was denied. 

The vehicle which we used is autogyro from [this post](http://discuss.px4.io/t/auto-g2-gyrocopter-with-px4-autopilot/9942/10).

bkueng, 2019-04-10:
From the log I don't see anything suspicious, except for high vibrations. Did you try with a different SD card or format it?

kaklik, 2019-04-10:
Yes, high vibrations are a problem. But I do not expect the vibrations could be able to stop logging on the SD card completely. 

We do not format the SD card or use another yet. 

In my opinion, the proper behavior of the system in the situation where is unable to log on the card is the following: 
 * It will try to create another file or retry to write to existing one.
 * Post a message about this situation via the telemetry

It seems nothing from the list has happened. Except there is a suspicuous situation where logger used the file name from previous day.

kaklik, 2019-04-20:
The format of SD card does not affect this behavior. I am able to reproduce the situation even with newly formatted SDcard. 

Additionally, we experienced a test fly where the log file was not created at all.

A deeper investigation of that situation shows it is especially easy to get the firmware to a situation where the log is not created. 
 
The one method follows: 

1. start logging from PX4 console by "logger on"
2. wobble with SDcard or remove it complettely
3. insert SDcard back
4. check "logger status" - there will be either "no logging", or last log file message. 

In both cases is not possible to start logging fuethermore.

[image]

Yes, it is some sort of hardware issue, but the firmware definitely needs to be resistant to this.

LorenzMeier, 2019-04-21:
The way open source works is that the best solutions come from the people most affected by an issue. The check if a file is writable is here: https://github.com/PX4/Firmware/blob/master/src/modules/logger/log_writer_file.cpp#L257

I recommend you connect a console (https://dev.px4.io/en/debug/system_console.html) and experiment a bit with various checks on the file system to see how you can build a check and alert for this.

kaklik, 2019-07-20:
The issue still lacks a solution.

julianoes, 2019-07-22:
> wobble with SDcard or remove it complettely

PX4 assumes that the SD card stays inserted and doesn't "wobble". If you could make a pull request to make logger more robust for this case that would be great.

redxeth, 2019-11-07:
Hi-- I've recently been having an issue with logger stopping mid-flight.

mhkabir, 2019-11-07:
@redxeth what software version are you running?

redxeth, 2019-11-07:
PX4 1.9.2 stable

davids5, 2019-11-07:
Master and (correct me if I am wrong) 1.10 is built with a SDMMC change in nuttx that fixed a bug where the calling thread hung indefinitely on a semaphore that would never get signaled.  Now the when the card fails, the calling code does not  hang but returns the error. The card is not accessible until the system is rebooted.  This has the same semantics as before but allows the caller to not hang.  This is critical to the dataman as it can can now RTL as apposed to be hung at a waypoint.

redxeth, 2019-11-07:
It's possible.  I found some wording in the Nuttx-7.28 release notes regarding MMCSD SDIO:

      - MMCSD SDIO:  The mmcsd_sdio driver makes calls for dcache
        invalidation through the chip specific architecture function
        SDIO_DMARECVSETUP(). I changed the arch/arm/stm32f7 chips to use
        arch_invalidate_dcache_by_addr() instead of arch_invalidate_dcache().

        This commit includes additional changes to mmcsd_sdio.c.  I
        created SDIO_DMADELYDINVLDT() (DMA delayed invalidate) to
        invalidate store-into mode dcaches after the DMA transfer.  I
        have been using SDIO_DMADELYDINVLDT() for several weeks now and
        it has fixed the problems that I previously reported regarding
        non-cache aligned buffer invalidation errors (for my store-through
        dcache). However, it does not permit use of unaligned DMA buffers
        for store-into mode dcaches.

Hard to tell which version of Nuttx PX4-1.10.0 was built with.

redxeth, 2019-11-11:
FYI everyone there is a new PR by @davids5 : https://github.com/PX4/Firmware/pull/13440

kaklik, 2022-07-18:
> FYI everyone there is a new PR by @davids5 : #13440

That seems to at least partially resolve that issue.

closed 2022-07-18 by kaklik