PX4/PX4-Autopilot #11824: Pixhawk Cube v2 Return to land failure

jorge789, 2019-04-09:
**Describe the bug**
- Modes Tested
Position Mode: Good.
Mission Plan Mode (Automated)
RTL (Return To Land

- Procedure
Arm and Take off in position mode, after flying for approximately one minute, switched to mission plan mode  behavior,the vehicle took-off and reached the designated altitude (35m) as expected; issue started once it reached the altitude of 35m, UAV began to oscillate and descend at the same time but the descend was not continuous, it had short pauses.

https://review.px4.io/plot_app?log=e9d7b116-487a-45c2-98f9-52d9cec4d88f

**Drone (please complete the following information):**
- Describe the type of drone.F450 Pixhawk Cube v2

Video:  https://www.youtube.com/watch?v=Sv2jUGdmNBo&feature=youtu.be

julianoes, 2019-07-10:
Sorry for the late response here. Thanks for the issue, log, and video, that's helpful!

Now I'm really puzzled at the weird thrust values as well as the pitch and roll setpoints. It looks like there were conflicting setpoints published somewhere that interfered with the controllers setpoints.

Is it possible that offboard messages were sent or QGC sent setpoints from a joystick?

julianoes, 2019-10-08:
Please re-open if this problem persists.

closed 2019-10-08 by julianoes