PX4/PX4-Autopilot #11830: Here 2 compass not working?

RomanBapst, 2019-04-10:
Tested with Pixhawk cube and latest master.

RomanBapst, 2019-04-10:
In then went on and tied with a here 1 GPS which worked.
But then I noticed that the external compass was MAG_3 and QGC did not show the option to set an external mag rotation.

julianoes, 2019-04-11:
I think this is a duplicate of #11230.

dagar, 2019-04-11:
Fixed in https://github.com/PX4/Firmware/pull/11837

dagar, 2019-04-11:
> But then I noticed that the external compass was MAG_3 and QGC did not show the option to set an external mag rotation.

Likely fixed by https://github.com/PX4/Firmware/pull/11838

dagar, 2019-04-11:
Support for the compass in the Here 2 GPS is now in master.https://github.com/PX4/Firmware/commit/95a0803b149770b8d1a5817611cd55e43a4ad296

closed 2019-04-11 by dagar